zephyrproject-rtos/zephyr #27166: tests/kernel/sched/schedule_api need add ram limitaion as some platform not support

hakehuang, 2020-07-27:
when build on lpcxpresso55s16_ns I see below error

`
[ 91%] Linking C executable zephyr_prebuilt.elf

Memory region         Used Size  Region Size  %age Used

           FLASH:       73948 B       246 KB     29.36%

            SRAM:       16884 B        32 KB     51.53%

        IDT_LIST:         152 B         2 KB      7.42%

[ 91%] Built target zephyr_prebuilt

Scanning dependencies of target obj_list

[ 92%] Generating kobject_hash.gperf

[ 92%] Built target obj_list

Scanning dependencies of target output_src_pre

[ 92%] Generating kobject_hash_preprocessed.c

[ 93%] Built target output_src_pre

[ 94%] Generating kobject_hash.c

Scanning dependencies of target output_lib

[ 94%] Building C object zephyr/CMakeFiles/output_lib.dir/kobject_hash.c.obj

[ 95%] Linking C static library liboutput_lib.a

[ 96%] Built target output_lib

Scanning dependencies of target output_obj_renamed

[ 96%] Generating kobject_hash_renamed.o

[ 96%] Built target output_obj_renamed

Scanning dependencies of target linker_pass_final_script_target

[ 97%] Generating linker_pass_final.cmd

[ 97%] Built target linker_pass_final_script_target

[ 98%] Generating isr_tables.c

Scanning dependencies of target zephyr_final

[ 98%] Building C object zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj

[100%] Building C object zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj

[100%] Linking C executable zephyr.elf

/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: zephyr.elf section `priv_stacks_noinit' will not fit in region `SRAM'

/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: region `SRAM' overflowed by 1296 bytes

collect2: error: ld returned 1 exit status

make[2]: *** [zephyr/zephyr.elf] Error 1

zephyr/CMakeFiles/zephyr_final.dir/build.make:143: recipe for target 'zephyr/zephyr.elf' failed

CMakeFiles/Makefile2:2263: recipe for target 'zephyr/CMakeFiles/zephyr_final.dir/all' failed

make[1]: *** [zephyr/CMakeFiles/zephyr_final.dir/all] Error 2

make: *** [all] Error 2

Makefile:102: recipe for target 'all' failed

script returned exit code 2
`

on hash 
ab3e778f47d

nashif, 2020-07-27:
do we really need an issue opened for this? many tests do not fit on many boards for various reasons. Just send the fix

hakehuang, 2020-07-28:
ok

closed 2020-07-28 by hakehuang