zephyrproject-rtos/zephyr #27174: TCP Server don't get the right data from the client

shiraez, 2020-07-27:
When i open tcp server and tcp client with different threads, buffer filled with trash using zsock_recv.

TCP Server:
```
int
server()
{
struct sockaddr_in addr;
    uint16_t port = 40000;
    char              *dst = "192.0.3.1";
    int               s, new_socket = -1;
    int                ret = -1;
    int addrlen = sizeof(addr);
    int opt = 1;
    char buffer[1024] = {0};
    char* messg = "Hello from server";

    memset(&addr, 0, sizeof(addr));
    addr.sin_family = AF_INET;
    addr.sin_addr.s_addr = INADDR_ANY;
    addr.sin_port = htons(port);
    if ((s = zsock_socket(AF_INET, SOCK_STREAM, 0)) < 0)
    {
        return -1;
    }

    if (zsock_setsockopt(s, SOL_SOCKET, SO_REUSEADDR, &opt, sizeof(opt)))
    {
        return -1;
    }

    if (zsock_bind(s, (struct sockaddr *)&addr,
                                 sizeof(addr))<0)
    {
        clicon_err(OE_UNIX, errno, "bind");
        return -1;
    }
    if (zsock_listen(s, 3) < 0)
    {
        return -1;
    }
    if ((new_socket = zsock_accept(s, (struct sockaddr *)&addr,
                       (socklen_t*)&addrlen))<0)
    {
        return -1;
    }
    ret = zsock_recv( new_socket , buffer, sizeof(buffer), 0);
    printk("len %d  %s\n",ret, buffer);
   return 0;

}
```

TCP client:
```
int
client()
{
    struct sockaddr_in addr;
    uint16_t port = 40000;
    char              *dst = "192.0.3.1";
    int               *s = -1;
    int                ret = -1;
    char* messg = "Hello from client";
    char buffer[1024] = {0};

    memset(&addr, 0, sizeof(addr));
    addr.sin_family = AF_INET;
    addr.sin_port = htons(port);

    if ((s = zsock_socket(addr.sin_family, SOCK_STREAM, 0)) < 0) {
    clicon_err(OE_CFG, errno, "socket");
    return -1;
    }
    if (zsock_inet_pton(addr.sin_family, dst, &addr.sin_addr) != 1)
    return -1;

    if (zsock_connect(s, (struct sockaddr*)&addr, sizeof(addr)) < 0){
    zsock_close(s);
    return -1;
    }
    if((ret = zsock_send(s , messg , strlen(messg) , 0 )) < 0)
      return -1;
    done:
    return 0;
}
```
**Environment**:
  -POSIX

jukkar, 2020-07-28:
In client you have this which is wrong.

`> int               *s = -1;`

After I removed `*`, the thing worked just fine.

shiraez, 2020-07-28:
Hi, thanks for your answer, but it doesn't work.

timork, 2020-07-28:
@ shiraez
Please provide better example for the issue

jukkar, 2020-07-28:
I used this test program for testing
https://github.com/jukkar/zephyr/tree/bug-27174-client-server-communication
See the code in `samples/net/sockets/client-server`

shiraez, 2020-07-28:
I ran your test but this is still print "len 1024" and the buffer is 0.
I'm using an older version of zephyr(114 I think), do you know if this is related to this problem?

jukkar, 2020-07-29:
When using latest upstream, I get
```
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.3.0-1351-g2a9a3bc926b6  ***

[00:00:00.000,000] <inf> net_config: Initializing network
[00:00:00.010,000] <inf> net_config: IPv4 address: 192.0.2.1
uart:~$ len 17  Hello from client
[00:00:00.110,000] <inf> net_config: IPv6 address: fe80::5054:ff:fe12:3456
[00:00:00.110,000] <inf> net_config: IPv6 address: fe80::5054:ff:fe12:3456
```
When running it in 1.14, I get the same print `len 17  Hello from client` but there is a NULL ptr access in TCP. This with qemu_x86 and native_posix. The is some issue that has been fixed in upstream but is missing in 1.14.
What board are you using?

shiraez, 2020-07-29:
I use native_posix 64 bit.

closed 2020-10-12 by github-actions[bot]

Anuhya0, 2024-08-27:
can you please help me out in my project on tcp client-server communication using cora z7 and h265