zephyrproject-rtos/zephyr #27177: Unable to build samples/bluetooth/st_ble_sensor for steval_fcu001v1 board

mtuxpe, 2020-07-27:
**Describe the bug**
Unable to build  sample bluetooth application st_ble_sensor  for steval_fcu001v1 board 

What have you tried to diagnose or workaround this issue?
I tried the same build for other boards and it worked without  any error:
board nrf52840dk_nrf52840:  successful build without error
board nucleo_wb55rg: successful build without error

**To Reproduce**
Steps to reproduce the behavior:
1. git status
On branch master
Your branch is up to date with 'origin/master'.
nothing to commit, working tree clean

2. rm -rf build
3. west build -b steval_fcu001v1 samples/bluetooth/st_ble_sensor
4. See error

**Expected behavior**
The build  for samples/bluetooth/st_ble_sensor should run without errors

**Impact**
Use the kernel as a development  platform for drones

**Logs and console output**
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/msam/zephyr_2.3.x/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/msam/zephyr_2.3.x/zephyrproject/zephyr/samples/bluetooth/st_ble_sensor
-- Zephyr version: 2.3.99 (/home/msam/zephyr_2.3.x/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.2") found components: Interpreter 
-- Board: steval_fcu001v1
-- Found toolchain: zephyr (/home/msam/zephyr-sdk)
-- Found west: /home/msam/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /home/msam/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/msam/zephyr_2.3.x/zephyrproject/zephyr/boards/arm/steval_fcu001v1/steval_fcu001v1.dts
-- Generated zephyr.dts: /home/msam/zephyr_2.3.x/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/msam/zephyr_2.3.x/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/msam/zephyr_2.3.x/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/msam/zephyr_2.3.x/zephyrproject/zephyr/boards/arm/steval_fcu001v1/steval_fcu001v1_defconfig'
Merged configuration '/home/msam/zephyr_2.3.x/zephyrproject/zephyr/samples/bluetooth/st_ble_sensor/prj.conf'
Configuration saved to '/home/msam/zephyr_2.3.x/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/msam/zephyr_2.3.x/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/msam/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/msam/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/msam/zephyr_2.3.x/zephyrproject/zephyr/build
-- west build: building application
[1/161] Preparing syscall dependency handling

[32/161] Building C object CMakeFiles/app.dir/src/button_svc.c.obj
FAILED: CMakeFiles/app.dir/src/button_svc.c.obj 
ccache /home/msam/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.3.0-1324-g1c2f127e3fd1 -DCORE_CM4 -DHSE_VALUE=16000000 -DKERNEL -DSTM32F401xC -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/msam/zephyr_2.3.x/zephyrproject/zephyr/samples/bluetooth -I/home/msam/zephyr_2.3.x/zephyrproject/zephyr/include -Izephyr/include/generated -I/home/msam/zephyr_2.3.x/zephyrproject/zephyr/soc/arm/st_stm32/stm32f4 -I/home/msam/zephyr_2.3.x/zephyrproject/zephyr/drivers -I/home/msam/zephyr_2.3.x/zephyrproject/zephyr/soc/arm/st_stm32/common -I/home/msam/zephyr_2.3.x/zephyrproject/zephyr/subsys/bluetooth -I/home/msam/zephyr_2.3.x/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/msam/zephyr_2.3.x/zephyrproject/modules/hal/stm32/stm32cube/stm32f4xx/soc -I/home/msam/zephyr_2.3.x/zephyrproject/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include -I/home/msam/zephyr_2.3.x/zephyrproject/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/Legacy -I/home/msam/zephyr_2.3.x/zephyrproject/modules/crypto/tinycrypt/lib/include -isystem /home/msam/zephyr_2.3.x/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/msam/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include -isystem /home/msam/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include-fixed -Os -imacros/home/msam/zephyr_2.3.x/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros/home/msam/zephyr_2.3.x/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/msam/zephyr_2.3.x/zephyrproject/zephyr/samples/bluetooth/st_ble_sensor=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/msam/zephyr_2.3.x/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/msam/zephyr_2.3.x/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/button_svc.c.obj -MF CMakeFiles/app.dir/src/button_svc.c.obj.d -o CMakeFiles/app.dir/src/button_svc.c.obj   -c /home/msam/zephyr_2.3.x/zephyrproject/zephyr/samples/bluetooth/st_ble_sensor/src/button_svc.c
In file included from /home/msam/zephyr_2.3.x/zephyrproject/zephyr/include/arch/arm/aarch32/arch.h:20,
                 from /home/msam/zephyr_2.3.x/zephyrproject/zephyr/include/arch/cpu.h:19,
                 from /home/msam/zephyr_2.3.x/zephyrproject/zephyr/include/kernel_includes.h:38,
                 from /home/msam/zephyr_2.3.x/zephyrproject/zephyr/include/kernel.h:17,
                 from /home/msam/zephyr_2.3.x/zephyrproject/zephyr/include/zephyr.h:18,
                 from /home/msam/zephyr_2.3.x/zephyrproject/zephyr/samples/bluetooth/st_ble_sensor/src/button_svc.c:17:
/home/msam/zephyr_2.3.x/zephyrproject/zephyr/samples/bluetooth/st_ble_sensor/src/button_svc.c: In function 'button_init':
/home/msam/zephyr_2.3.x/zephyrproject/zephyr/include/devicetree.h:205:32: error: 'DT_N_ALIAS_sw0_P_gpios_IDX_0_PH_P_label' undeclared (first use in this function)
  205 | #define DT_ALIAS(alias) DT_CAT(DT_N_ALIAS_, alias)
      |                                ^~~~~~~~~~~
/home/msam/zephyr_2.3.x/zephyrproject/zephyr/include/devicetree.h:1786:38: note: in definition of macro 'DT_CAT'
 1786 | #define DT_CAT(node_id, prop_suffix) node_id##prop_suffix
      |                                      ^~~~~~~
[... 469 lines cut ...]
      |                           ^~~~~~~~~~~
/home/msam/zephyr_2.3.x/zephyrproject/zephyr/include/logging/log.h:40:25: note: in expansion of macro 'Z_LOG'
   40 | #define LOG_ERR(...)    Z_LOG(LOG_LEVEL_ERR, __VA_ARGS__)
      |                         ^~~~~
/home/msam/zephyr_2.3.x/zephyrproject/zephyr/samples/bluetooth/st_ble_sensor/src/button_svc.c:90:3: note: in expansion of macro 'LOG_ERR'
   90 |   LOG_ERR("Error %d: failed to configure interrupt on pin "
      |   ^~~~~~~
/home/msam/zephyr_2.3.x/zephyrproject/zephyr/samples/bluetooth/st_ble_sensor/src/button_svc.c:91:10: note: format string is defined here
   91 |    "%d '%s'\n", ret, BUT_PIN, DT_LABEL(DT_ALIAS(sw0)));
      |         ~^
      |          |
      |          char *
      |         %ln
[41/161] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/clock_control/clock_stm32_ll_common.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/msam/zephyr_2.3.x/zephyrproject/zephyr/build

**Environment (please complete the following information):**

 - OS: Linux Ubuntu 20.04.1 LTS
 - Toolchain :Zephyr SDK 0.11.4
 - Commit  1c2f127e3fd1a4aa49dfe30e06f992655d5f5c38

**Additional context**
Add any other context about the problem here.

mtuxpe, 2020-07-27:
This error caused by the absence of a button on  steval_fcu001v1 board.
Thanks thanks for the tip  Brenton Chetty .Closing issue.

closed 2020-07-27 by mtuxpe

ASL07, 2022-05-23:
Hi, what was the fix for this? I am seeing the same issue in native_posix board

as-iotex, 2022-05-23:
Hi, what was the fix for this? I am seeing the same issue in native_posix board