zephyrproject-rtos/zephyr #27181: New drivers out of device tree

thomasoggiersoprod, 2020-07-27:
Hello,
I try to add new customs drivers in my project out of the device tree but I can't do it.
I've add in my project folder/files:
1 the file prj.conf
  - add line
        CONFIG_GPIO_MCP23S08=y
2 the file drivers/CMakeLists.txt
  - add line
        add_subdirectory_ifdef(CONFIG_MCP23S08 mcp23s08)
3 the file drivers/KConfig
  - add line
        rsource "mcp23s08/Kconfig"
4 the folder drivers/mcp23s08
5 the file drivers/mcp23s08/CMakeLists.txt
  - add line
        zephyr_library_sources_ifdef(CONFIG_GPIO_MCP23S08 gpio_mcp23s08.c)
6 the file drivers/mcp23s08/KConfig
  - add line
        menuconfig gpio_mcp23s08
        bool "gpio_mcp23s08 driver"
        default y
        select BLAH
        depends on MEH
        help
         Enable support for gpio_mcp23s08 driver
        if gpio_mcp23s08
        module = gpio_mcp23s08
        module-str = gpio_mcp23s08
        source "subsys/logging/Kconfig.template.log_config"
        endif
7 the file module.yml
  - add line
        build:
        cmake: .
        kconfig: Kconfig
8 the files gpio_mcp23s08.c/.h
9 the folder dts/bindings/gpio
10 the file dts/bindings/gpio/microchip,mcp23s08.yaml

But when I try to compile, I get the error:
-> warning: attempt to assign the value 'y' to the undefined symbol GPIO_MCP23S08

I've allready tried my driver in "driver in tree" and it works.
Maybe you can help me, I can't find any real documentation/sample.
thanks
Thomas Oggier

tejlmand, 2020-08-06:
@thomasoggiersoprod the following is based on my assumption of you folder layout.

It seems you are trying to place you out of tree device as a Zephyr module: https://docs.zephyrproject.org/latest/guides/modules.html

So your structure might look similar to:
```
<module-path>
| --- zephyr
|       |--- module.yml
|
|--- drivers
        |--- CMakeLists.txt
        |--- Kconfig
```

In that case, you should either have the `zephyr/module.yml` with:
```
build:
  cmake: drivers
  kconfig: drivers/Kconfig
```
or place a CMakeLists.txt and Kconfig file at `<module-path>` root, that include the files in `drivers`, so you would have:
```
<module-path>
| --- zephyr
|       |--- module.yml
|--- CMakeLists.txt
|--- Kconfig
```

also remember, that if you want to include modules, that is not in a repository specified in `west.yml`, then you must use:
```
cmake -DZEPHYR_EXTRA_MODULES=<module-path>  .....
```

thomasoggiersoprod, 2020-08-06:
I think will are not talking about the same folder layout, so I set it in attachment.

[app01.zip](https://github.com/zephyrproject-rtos/zephyr/files/5035532/app01.zip)

tejlmand, 2020-08-06:
I see one mistake.

In app01/CMakeLists.txt:13:
```
add_subdirectory(${CMAKE_CURRENT_SOURCE_DIR}/drivers)
```

but you also have:
```
build:
  cmake: drivers
  kconfig: drivers/Kconfig
```

which will add the same drivers folder to the build system.
Thus, I become a bit unsure exactly how the system will behave, but you ought to see some errors if multiple libraries with the same name are defined.

So I currently suspect the module in not loaded into the build system.

Could you check the content of:
```
<build-folder>/zephyr_modules.txt 
```

in there you should see a line similar to:
```
"app01":"<patrh-to>/app01"
```

also check  the output of `west list` to see `app01` is correctly displayed.

thomasoggiersoprod, 2020-08-07:
So in my "<build-folder>/zephyr_modules.txt " I get `"mcuboot":"/home/thomas/Documents/IoT_Hevs/soprod_iot/bootloader/mcuboot/zephyr"
"mcumgr":"/home/thomas/Documents/IoT_Hevs/soprod_iot/modules/lib/mcumgr"
"nrfxlib":"/home/thomas/Documents/IoT_Hevs/soprod_iot/nrfxlib"
"nrf":"/home/thomas/Documents/IoT_Hevs/soprod_iot/nrf"
"cmsis":"/home/thomas/Documents/IoT_Hevs/soprod_iot/modules/hal/cmsis"
"nordic":"/home/thomas/Documents/IoT_Hevs/soprod_iot/modules/hal/nordic"
"mbedtls":"/home/thomas/Documents/IoT_Hevs/soprod_iot/modules/crypto/mbedtls"
"segger":"/home/thomas/Documents/IoT_Hevs/soprod_iot/modules/debug/segger"
"tinycbor":"/home/thomas/Documents/IoT_Hevs/soprod_iot/modules/lib/tinycbor/zephyr"
"tinycrypt":"/home/thomas/Documents/IoT_Hevs/soprod_iot/modules/crypto/tinycrypt"
"nrf_hw_models":"/home/thomas/Documents/IoT_Hevs/soprod_iot/modules/bsim_hw_models/nrf_hw_models"`
and with the cmd "west list" I get `manifest     app01                        HEAD                                     N/A
zephyr       zephyr                       v2.3.0-rc1-ncs1                          https://github.com/nrfconnect/sdk-zephyr
mcuboot      bootloader/mcuboot           v1.6.0-rc2-ncs1                          https://github.com/nrfconnect/sdk-mcuboot
mcumgr       modules/lib/mcumgr           v0.0.1-ncs2                              https://github.com/nrfconnect/sdk-mcumgr
nrfxlib      nrfxlib                      v1.3.0                                   https://github.com/nrfconnect/sdk-nrfxlib
mbedtls-nrf  mbedtls                      mbedtls-2.16.6                           https://github.com/ARMmbed/mbedtls
nrf          nrf                          v1.3.0                                   https://github.com/nrfconnect/sdk-nrf
cmsis        modules/hal/cmsis            542b2296e6d515b265e25c6b7208e8fea3014f90 https://github.com/zephyrproject-rtos/cmsis
hal_nordic   modules/hal/nordic           c84263d8ecc7767b25d9abab9502c9e748d045e9 https://github.com/zephyrproject-rtos/hal_nordic
mbedtls      modules/crypto/mbedtls       821154171b246f64eaeef3ccc267f58d8274739a https://github.com/zephyrproject-rtos/mbedtls
segger       modules/debug/segger         6fcf61606d6012d2c44129edc033f59331e268bc https://github.com/zephyrproject-rtos/segger
tinycbor     modules/lib/tinycbor         40daca97b478989884bffb5226e9ab73ca54b8c4 https://github.com/zephyrproject-rtos/tinycbor
tinycrypt    modules/crypto/tinycrypt     3e9a49d2672ec01435ffbf0d788db6d95ef28de0 https://github.com/zephyrproject-rtos/tinycrypt
nrf_hw_models modules/bsim_hw_models/nrf_hw_models fec69703cb1ca06fcdab6d5fde01274f0fc5c759 https://github.com/zephyrproject-rtos/nrf_hw_models`

If I remove the line 13 add_subdirectory... or the "zephyr/module.yml" file, I get each time the same result

tejlmand, 2020-08-07:
I just noticed one thing that could be a root cause.

In your `<manifest-repo>/west.yml` there is a section like:
```
  self:
    path: app01
```

which is revealed by this line:
```
manifest app01 HEAD N/A
```

this means that for this to work, your folder structure should be:
```
/home/thomas/Documents/IoT_Hevs/soprod_iot/app01/...
```

but if this code is actually located in another folder, let's say:
```
/home/thomas/Documents/IoT_Hevs/soprod_iot/myapp/...
```
then things will not work as expected.

Could you check up on that.
And also provide the output from:
```
west config manifest.path
```

thomasoggiersoprod, 2020-08-07:
I'm not sur to understand what you are talking about.

All my folder are in the folder /home/thomas/Documents/IoT_Hevs/soprod_iot/app01/
[image]

 and the result of the command west config manifest.path is "app01"

tejlmand, 2020-08-07:
@thomasoggiersoprod  found the issue.
You are using `zephyr/modules.yml` which is wrong.
The correct name is: `zephyr/module.yml` without `s`.

thomasoggiersoprod, 2020-08-07:
Thanks, stupid mistake.
When I change it, the files zephyr_modules.txt add this line, great.
`"app01":"/home/thomas/Documents/IoT_Hevs/soprod_iot/app01/drivers"`
But I get this compilation error:
`thomas@thomas-HP:~/Documents/IoT_Hevs/soprod_iot/app01$ west build -p always -b nrf9160dk_nrf9160ns
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/thomas/Documents/IoT_Hevs/soprod_iot/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/thomas/Documents/IoT_Hevs/soprod_iot/app01
-- Zephyr version: 2.3.0-rc1 (/home/thomas/Documents/IoT_Hevs/soprod_iot/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.2") found components: Interpreter 
-- Board: nrf9160dk_nrf9160ns
-- Found toolchain: zephyr (/home/thomas/zephyr-sdk)
-- Found west: /home/thomas/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /home/thomas/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/thomas/Documents/IoT_Hevs/soprod_iot/zephyr/boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160ns.dts
-- Found devicetree overlay: /home/thomas/Documents/IoT_Hevs/soprod_iot/app01/nrf9160dk_nrf9160ns.overlay
nrf9160dk_nrf9160ns.dts.pre.tmp:394.32-398.5: Warning (simple_bus_reg): /soc/cryptocell-sw: missing or empty reg/ranges property
nrf9160dk_nrf9160ns.dts.pre.tmp:93.42-105.3: Warning (unique_unit_address_if_enabled): /soc/peripheral@40000000/flash-controller@39000: duplicate unit-address (also used in node /soc/peripheral@40000000/kmu@39000)
nrf9160dk_nrf9160ns.dts.pre.tmp:342.19-348.3: Warning (unique_unit_address_if_enabled): /soc/peripheral@40000000/clock@5000: duplicate unit-address (also used in node /soc/peripheral@40000000/power@5000)
-- Generated zephyr.dts: /home/thomas/Documents/IoT_Hevs/soprod_iot/app01/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/thomas/Documents/IoT_Hevs/soprod_iot/app01/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/thomas/Documents/IoT_Hevs/soprod_iot/app01/Kconfig
Loaded configuration '/home/thomas/Documents/IoT_Hevs/soprod_iot/zephyr/boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160ns_defconfig'
Merged configuration '/home/thomas/Documents/IoT_Hevs/soprod_iot/app01/prj.conf'
Configuration saved to '/home/thomas/Documents/IoT_Hevs/soprod_iot/app01/build/zephyr/.config'
Kconfig header saved to '/home/thomas/Documents/IoT_Hevs/soprod_iot/app01/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/thomas/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/thomas/.cache/zephyr
Changed board to secure nrf9160dk_nrf9160 (NOT NS)

=== child image spm - nrf9160dk_nrf9160 begin ===
Including boilerplate (Zephyr base): /home/thomas/Documents/IoT_Hevs/soprod_iot/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/thomas/Documents/IoT_Hevs/soprod_iot/nrf/samples/nrf9160/spm
-- Zephyr version: 2.3.0-rc1 (/home/thomas/Documents/IoT_Hevs/soprod_iot/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.2") found components: Interpreter 
-- Board: nrf9160dk_nrf9160
-- Found toolchain: zephyr (/home/thomas/zephyr-sdk)
-- Found west: /home/thomas/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /home/thomas/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/thomas/Documents/IoT_Hevs/soprod_iot/zephyr/boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160.dts
-- Found devicetree overlay: /home/thomas/Documents/IoT_Hevs/soprod_iot/nrf/samples/nrf9160/spm/nrf9160dk_nrf9160.overlay
nrf9160dk_nrf9160.dts.pre.tmp:95.42-107.3: Warning (unique_unit_address_if_enabled): /soc/peripheral@50000000/flash-controller@39000: duplicate unit-address (also used in node /soc/peripheral@50000000/kmu@39000)
nrf9160dk_nrf9160.dts.pre.tmp:344.19-350.3: Warning (unique_unit_address_if_enabled): /soc/peripheral@50000000/clock@5000: duplicate unit-address (also used in node /soc/peripheral@50000000/power@5000)
-- Generated zephyr.dts: /home/thomas/Documents/IoT_Hevs/soprod_iot/app01/build/spm/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/thomas/Documents/IoT_Hevs/soprod_iot/app01/build/spm/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/thomas/Documents/IoT_Hevs/soprod_iot/zephyr/Kconfig
Loaded configuration '/home/thomas/Documents/IoT_Hevs/soprod_iot/zephyr/boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160_defconfig'
Merged configuration '/home/thomas/Documents/IoT_Hevs/soprod_iot/nrf/samples/nrf9160/spm/prj.conf'
Merged configuration '/home/thomas/Documents/IoT_Hevs/soprod_iot/nrf/samples/nrf9160/spm/boards/nrf9160dk_nrf9160.conf'
Configuration saved to '/home/thomas/Documents/IoT_Hevs/soprod_iot/app01/build/spm/zephyr/.config'
Kconfig header saved to '/home/thomas/Documents/IoT_Hevs/soprod_iot/app01/build/spm/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/thomas/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/thomas/.cache/zephyr
-- Configuring done
-- Generating done
[... cut ...]
           FLASH:       27208 B       976 KB      2.72%
            SRAM:        6800 B       128 KB      5.19%
        IDT_LIST:          88 B         2 KB      4.30/home/thomas/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: Undefined initialization levels used.
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/thomas/Documents/IoT_Hevs/soprod_iot/app01/build`

tejlmand, 2020-08-07:
When trying to build like you, I can reproduce same error which seems to originate from the generated linker file in `<build-folder>/zephyr/linker.cmd`

```
_image_rodata_start = .;                                                                           
 initlevel :                                                                                        
 {                                                                                                  
  __init_start = .; __init_PRE_KERNEL_1_start = .; KEEP(*(SORT(.init_PRE_KERNEL_1[0-9]))); KEEP(*(SORT(.init_PRE_KERNEL_1[1-9][0-9]))); __init_PRE_KERNEL_2_start = .; KEEP(*(SORT(.init_PRE_KERNEL_2[0-9]))); KEEP(*(SORT(.init_PRE_KERNEL_2[1-9][0-9]))); __init_POST_KERNEL_start = .; KEEP(*(SORT(.init_POST_KERNEL[0-9]))); KEEP(*(SORT(.init_POST_KERNEL[1-9][0-9]))); __init_APPLICATION_start = .; KEEP(*(SORT(.init_APPLICATION[0-9]))); KEEP(*(SORT(.init_APPLICATION[1-9][0-9]))); __init_SMP_start = .; KEEP(*(SORT(.init_SMP[0-9]))); KEEP(*(SORT(.init_SMP[1-9][0-9]))); __init_end = .;
 } > FLASH                                                                                          
 sw_isr_table :                                                                                     
 {                                                                                                  
  . = ALIGN(0);                                                                                     
  *(.gnu.linkonce.sw_isr_table)                                                                     
 } > FLASH                                                                                          
 initlevel_error :                                                                                  
 {                                                                                                  
  KEEP(*(SORT(.init_[_A-Z0-9]*)))                                                                   
 }                                                                                                  
 ASSERT(SIZEOF(initlevel_error) == 0, "Undefined initialization levels used.")  
```

however, i'm not sure on the root cause of this.

thomasoggiersoprod, 2020-08-10:
Did you find a way to solve this problem?

tejlmand, 2020-08-10:
another mistake, when referring to Kconfig variables, then be sure to use the right name.

You are using `CONFIG_GPIO_mcp23s08_INIT_PRIORITY` instead of `CONFIG_GPIO_MCP23S08_INIT_PRIORITY` in `gpio_mcp23s08.c`

closed 2020-08-10 by tejlmand

thomasoggiersoprod, 2020-08-12:
Ok thanks and sorry, but now I can compile, but the device_get_binding result function is the same -> null...

tejlmand, 2020-08-14:
@ioannisg  is this something in your area of expertise ^^^