zephyrproject-rtos/zephyr #27182: USBD_STRING_DESCR_DEFINE ordering for custom interface

WaeCo, 2020-07-27:
We have a use case that requires our HID application to have an iInterface descriptor string assigned.

However defining an additional string descriptor using 
```
USBD_STRING_DESCR_DEFINE(primary) struct usb_hid_interface_descriptor interface_descr = { ... }
``` 
will mess up the order of the descriptor list. 
This is because static initialization order is undefined and in our case the newly defined descriptor will be placed first in the USB descriptor list. However the subsystem expects the first index to be the language descriptor and fails during initialization. (See usb_descriptor.c:430)

As a workaround we found that using `USBD_STRING_DESCR_DEFINE(primary.4)` works as the sections are sorted alphabetically.

A proper solution could be to expose an additional macro e.g. *USBD_CUSTOM_STRING_DESCR_DEFINE* which gets placed between the *USBD_TERM_DESCR_DEFINE* and *USBD_STRING_DESCR_DEFINE*

Additionally (this could be another issue) is a HID iInterface string descriptor a common use case? 
I could provide a pr that adds an optional *CONFIG_USB_HID_INTERFACE_DESC* to the HID class.

jfischer-phytec-iot, 2020-08-05:
>As a workaround we found that using USBD_STRING_DESCR_DEFINE(primary.4) works as the sections are sorted alphabetically.

If this workaround is acceptable for you then it is the way to go until rework of the stack was finished.

>Additionally (this could be another issue) is a HID iInterface string descriptor a common use case?

The use case, I guess, is to differentiate between multiple similar HID interfaces?

WaeCo, 2020-08-06:
> The use case, I guess, is to differentiate between multiple similar HID interfaces?

Yes, the report descriptor is the same (basically one large buffer) and then the driver on the host differentiates via the interface descriptor with application protocol to use. This is to stay backwards compatible to older driver versions which we can not change.

jfischer-no, 2023-07-26:
Resolved by bd4ddec0c8c822bbdd420bd558b62c1d1a532c16.
An interface string descriptor for the HID implementation is not planned in the current USB device stack.

closed 2023-07-26 by jfischer-no