zephyrproject-rtos/zephyr #27189: syscall stack usage report

andrewboie, 2020-07-27:
**Is your enhancement proposal related to a problem? Please describe.**
Syscalls run on fixed-sized privilege elevation stacks. There is a hard limit on how much stack space a system call can use because of this. We don't have good data on how much stack syscalls actually use. 

**Describe the solution you'd like**
I'd like to be able to use CONFIG_STACK_ANALYSIS to be able to generate a report, for every API exposed with __syscall, the upper bound on stack usage. Outliers that exceed the default privilege stack size should be flagged as issues.

This usage data must be from the syscall perspective; the measurements have to start from the syscall trap handler, through the verification and marshalling functions. We could probably get by with starting measurements at each `mrsh` function for every syscall.

nashif, 2020-07-27:
you mean CONFIG_STACK_USAGE i guess

andrewboie, 2020-07-27:
> you mean CONFIG_STACK_USAGE i guess

yes, derp

praton1729, 2023-05-18:
Hi guys, I would like to work on this. Does the zephyr roadmap still considers this a required feature?