zephyrproject-rtos/zephyr #27191: DAC features and API enhancements

martinjaeger, 2020-07-28:
This issue will be used to collect findings from DAC peripheral PRs and issues  which could influence a future update of the currently experimental API.

## Overview

### Supported DACs

- ST STM32L0
- NXP Kinetis MCUX

### Work in progress

- [Microchip SAM0](https://github.com/zephyrproject-rtos/zephyr/pull/26649)
- [TI DACx0508](https://github.com/zephyrproject-rtos/zephyr/pull/26040)

## Feature enhancements

### Enable/disable analog output buffer

This has been discussed already in the [initial PR for the DAC](https://github.com/zephyrproject-rtos/zephyr/pull/21805). It was decided to keep this out of the API configuration until we know how common a configurable output buffer is.

### Voltage references and gain

Voltage reference selection was kept out of initial API because a common implementation covering every MCU type was not obvious. [PR 26649](https://github.com/zephyrproject-rtos/zephyr/pull/26649) suggests a way using [new enum feature](https://github.com/zephyrproject-rtos/zephyr/issues/21273) in devicetree.

There are also DACs which support to select different voltage references per channel, e.g. [the external TI DACx0508](https://github.com/zephyrproject-rtos/zephyr/pull/26040).

Above mentioned DACx0508 series also supports to select a different gain (2, 1 or 0.5) for each channel. At least a gain <1 has a similar result as choosing a different voltage reference, so this might be combined in the API in some way.

### Resolution

The resolution is currently not configured via devicetree, but in `struct dac_channel_cfg`. The reason is that some DACs (e.g. in STM32) support to use lower than max. resolution for increased speed (similar to many ADCs). Only the max. resolution is a hardware feature, but the actual resolution can be configurable by the application.

It could make sense to add the resolution as an additional cell to the io-channels property, which currently only has one cell to select the channel. (see bindings [here](https://github.com/zephyrproject-rtos/zephyr/tree/master/dts/bindings/dac))

## Datasheets

### Internal DACs

- ST [STM32L0x2](https://www.st.com/resource/en/reference_manual/dm00108281-ultralowpower-stm32l0x2-advanced-armbased-32bit-mcus-stmicroelectronics.pdf)
- NXP Kinetis MCUX
- Microchip [SAM D21](https://ww1.microchip.com/downloads/en/DeviceDoc/SAM_D21_DA1_Family_DataSheet_DS40001882F.pdf)

### External DACs (I2C or SPI)

- Texas Instruments [DACx0508](https://www.ti.com/lit/ds/symlink/dac60508.pdf)

towen, 2022-11-21:
Currently, it is not possible to set up a DAC as internally-connected to another peripheral (COMP, OPAMP, for example) because pinctrl-0 is a required property in devicetree. 
Also, it would be useful to enable the sawtooth-generation features of the STM32G4 series DACs for power electronics applications
Related issue: #38629

martinjaeger, 2022-11-22:
Regarding the sawtooth generation we recently had a discussion to extend the DAC API: https://github.com/zephyrproject-rtos/zephyr/pull/51271#issuecomment-1303061312

This should be fairly straightforward. I will try to come up with a proposal for the API change in the next days.

Connecting the DAC to other internal peripherals would be more difficult, as we would have to describe this connection in Devicetree in some way. Maybe we could actually use something similar to pinctrl for internal connections.

Light411, 2022-11-22:
The elephant in the room IMO is the lack of proper support of the interconnection matrix for STM32 devices. That's a big piece, but it would help a lot, especially for power electronics application, or any other applications that involves having analog routine, or PWM routines independant from the MCU calculation.

KozhinovAlexander, 2025-04-12:
@martinjaeger Is there any progress?

martinjaeger, 2025-04-14:
> [@martinjaeger](https://github.com/martinjaeger) Is there any progress or solution already?

Not really. Do you have any specific need?

KozhinovAlexander, 2025-04-14:
> > [@martinjaeger](https://github.com/martinjaeger) Is there any progress or solution already?
> 
> Not really. Do you have any specific need?

Yes, connecting opamp to adc on stm32 for example, if you as about specific need. Or speaking more general: one peripheral shall pass it's results to the other independently of MCU vendor specific implementation.
I know, it could be solved at the moment for stm32 in particular, but not for all mentioned SoC in the description.

martinjaeger, 2025-04-14:
At least for STM32 it's possible to configure the DAC output as "internally connected" (see https://github.com/zephyrproject-rtos/zephyr/commit/834cc98f3919fca5eee0ac7930984da33a82f202). But it will still require some manual work to define the "consumer" of the DAC output.

KozhinovAlexander, 2025-04-17:
> At least for STM32 it's possible to configure the DAC output as "internally connected" (see https://github.com/zephyrproject-rtos/zephyr/commit/834cc98f3919fca5eee0ac7930984da33a82f202). But it will still require some manual work to define the "consumer" of the DAC output.

Have you thought/heard about interconnect  subsystem for zephyr?
https://www.kernel.org/doc/html/v5.2/interconnect/interconnect.html?hl=de-DE