zephyrproject-rtos/zephyr #27201: tests/kernel/sched/schedule_api fails on lpcxpresso54114_m4 with '-DCONF_FILE=prj_multiq.conf' '-DCONFIG_TIMESLICING=y'

hakehuang, 2020-07-28:
**Describe the bug**
2API failure on tests/kernel/sched/schedule_api config 
Platform:
lpcxpresso54114_m4
Location:
tests/kernel/sched/schedule_api 
configs:
'-DCONF_FILE=prj_multiq.conf' '-DCONFIG_TIMESLICING=y' 

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=lpcxpresso54114_m4 -DCONF_FILE=prj_multiq.conf' '-DCONFIG_TIMESLICING=y ..
3. make flash
4. See error

**Expected behavior**
test pass

**Impact**
kernel scheduler

**Logs and console output**

```
Predicted Cause: [Contain-failPatterns] ['PROJECT EXECUTION FAILED'] [... truncated '49' lines log ...] C Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false D Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false E Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false F Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false G Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false H Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false I Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false A Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false FAIL
 - test_slice_scheduling =================================================================== START - test_priority_scheduling
 ABCDEFGHI ABCDEFGHI ABCDEFGHI ABCDEFGHI ABCDEFGHI PASS - test_priority_scheduling =================================================================== START - test_wakeup_expired_timer_thread PASS - test_wakeup_expired_timer_thread =================================================================== START - test_user_k_wakeup PASS - test_user_k_wakeup =================================================================== START - test_user_k_is_preempt PASS - test_user_k_is_preempt =================================================================== Test suite threads_scheduling failed. =================================================================== 
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA:  ab3e778f47d

hakehuang, 2020-07-28:
@MaureenHelm, @dlech

galak, 2020-07-28:
Did some git bisection and tracked this down to commit 88765ad3282ce830bda8745b409538a0bcdeaf6a.

It seems reducing the number of char's printed somehow impacts this test.

MaureenHelm, 2020-09-18:
This doesn't reproduce in master (ae8dd148874a1f4a71a7d0b7de8acb22dd184de3) anymore. Bisecting shows it was fixed by 379b93f0d327916cce2f4e2ec521ce0c098cccf0

cc: @nashif @andyross

nashif, 2020-09-18:
> . Bisecting shows it was fixed by [379b93f](https://github.com/zephyrproject-rtos/zephyr/commit/379b93f0d327916cce2f4e2ec521ce0c098cccf0)

hmm, not sure how this would have fixed it.

closed 2020-09-25 by hakehuang