zephyrproject-rtos/zephyr #27202: tests/kernel/threads/thread_apis failure on lpcxpresso55s16_ns

hakehuang, 2020-07-28:
**Describe the bug**
tests/kernel/threads/thread_apis failure on lpcxpresso55s16_ns

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=lpcxpresso55s16_ns ..
3. make flash
4. See error

**Expected behavior**
test PASS

**Impact**
thread function

**Logs and console output**

```
  [... truncated '22' lines log ...]
    ===================================================================
    START - test_threads_priority_set
     PASS - test_threads_priority_set
    ===================================================================
    START - test_threads_abort_self
     PASS - test_threads_abort_self
    ===================================================================
    START - test_threads_abort_others
     PASS - test_threads_abort_others
    ===================================================================
    START - test_threads_abort_repeat
     PASS - test_threads_abort_repeat
    ===================================================================
    START - test_abort_handler
    Child threadCalling abort of child from parent
    Child thread's abort handler called
     PASS - test_abort_handler
    ===================================================================
    START - test_delayed_thread_abort
     PASS - test_delayed_thread_abort
    ===================================================================
    START - test_essential_thread_operation
     PASS - test_essential_thread_operation
    ===================================================================
    START - test_systhreads_main
     PASS - test_systhreads_main
    ===================================================================
    START - test_systhreads_idle
     PASS - test_systhreads_idle
    ===================================================================
    START - test_customdata_get_set_coop
     PASS - test_customdata_get_set_coop
    ===================================================================
    START - test_customdata_get_set_preempt
     PASS - test_customdata_get_set_preempt
    ===================================================================
    START - test_k_thread_foreach
    TEST: Newly added thread found
    TEST: tid: 0x200005e0, prio: 0
     PASS - test_k_thread_foreach
    ===================================================================
    START - test_thread_name_get_set
     PASS - test_thread_name_get_set
    ===================================================================
    START - test_thread_name_user_get_set
    ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:953
    	ESF could not be retrieved successfully. Shall never occur.
    ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:953
    	ESF could not be retrieved successfully. Shall never occur.
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA:  ab3e778f47d

hakehuang, 2020-07-28:
@MaureenHelm @dleach02

MaureenHelm, 2020-07-28:
@henrikbrixandersen could you please have a look at this? I don't currently have this board

galak, 2020-07-28:
@hakehuang just to verify are you running the "non-secure" board variant, and if so what is running as the trusted firmware side?

henrikbrixandersen, 2020-07-28:
> @hakehuang just to verify are you running the "non-secure" board variant, and if so what is running as the trusted firmware side?

The Zephyr board definition for this board only support NS for now.

hakehuang, 2020-07-29:
> @hakehuang just to verify are you running the "non-secure" board variant, and if so what is running as the trusted firmware side?

nothing, I leave the trusted firmware blank.

henrikbrixandersen, 2020-07-29:
@hakehuang I can reproduce the error here.

ioannisg, 2020-08-10:
@hakehuang @henrikbrixandersen I guess, then, this should not be supported; running a firmware built with the _ns version of the board _requires_ a trusted firmware counterpart be present. If not the _ns should have never been used.

Note that if you would simply want to ignore TrustZone then you need to use the default board name (without the _ns).

hakehuang, 2020-08-13:
close this issue, and document this . thanks @ioannisg

closed 2020-08-13 by hakehuang