zephyrproject-rtos/zephyr #27203: tests/subsys/storage/flash_map failure on twr_ke18f

hakehuang, 2020-07-28:
**Describe the bug**
platform: 
twrke18f
Location:
              tests/subsys/storage/flash_map 

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=twr_ke18f ..
3. make flash
4. See error

**Expected behavior**
test pass

**Impact**
flash usage

**Logs and console output**

```
    *** Booting Zephyr OS version 2.3.99  ***
    Running test suite test_flash_map
    ===================================================================
    START - test_flash_area_get_sectors
    
        Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/storage/flash_map/src/main.c:96: test_flash_area_get_sectors: (rc == 0 is false)
    area not erased
     FAIL - test_flash_area_get_sectors
    ===================================================================
    Test suite test_flash_map failed.
    ===================================================================
    PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA:  ab3e778f47d

hakehuang, 2020-07-28:
@MaureenHelm  @dlech

henrikbrixandersen, 2020-07-29:
If I do a full chip erase (`pyocd erase --chip`) prior to running this test, the test passes:

```
*** Booting Zephyr OS build zephyr-v2.3.0-1372-g5e2e7a0fcf6f  ***
Running test suite test_flash_map
===================================================================
START - test_flash_area_get_sectors
 PASS - test_flash_area_get_sectors
===================================================================
Test suite test_flash_map succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

```

Not sure this is specific to the twr_ke18f. @nvlsianpu Any input on this?

dleach02, 2020-07-29:
@henrikbrixandersen, did you see the test fail before doing the full chip erase?

I've seen this before and would really like to figure out what sequence gets us to this state.

MaureenHelm, 2020-07-29:
> If I do a full chip erase (`pyocd erase --chip`) prior to running this test

Btw, see 8c7cef6da0175f1f6996a54dfe46adb79e6b48b6

henrikbrixandersen, 2020-07-29:
> @henrikbrixandersen, did you see the test fail before doing the full chip erase?
> 
> I've seen this before and would really like to figure out what sequence gets us to this state.

@dleach02 Yes, it was failing just before I did the full chip erase. I will try to see if I can get it back into the failing state.

dleach02, 2020-08-13:
I got one of these boards in and was able to reproduce the reported problem. But it doesn't seem to happen all the time. I personally don't believe it erasing the whole flash before running this test should have any bearing on the results as the test case does the following:

1) erases the flash block used for the test
2) writes a 256byte block of 0xa5 to the beginning and end of every sector. Verifies each write with a read of the block. Sector size is 0x1000.
3) erases the flash block again
4) tests to see if the flash block is erased.

When I got the error I added some code to see where it failed and I got the following offsets:

1) 0x1f000
2) 0x31f00
3) 0x31f00

Then no more failures. Note that it actually passed a number of times between the second and third failures.

The block dump of the two failures  were the same with:

zork: 0x31f00, rc: 90

ff ff ff ff ff ff ff ff - ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff - ff ff ff ff ff ff ff ff
a5 a5 a5 a5 a5 a5 a5 a5 - a5 a5 a5 a5 a5 a5 a5 a5
ff ff ff ff ff ff ff ff - ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff - ff ff ff ff ff ff ff ff
a5 a5 a5 a5 a5 a5 a5 a5 - a5 a5 a5 a5 a5 a5 a5 a5
a5 a5 a5 a5 a5 a5 a5 a5 - a5 a5 a5 a5 a5 a5 a5 a5
ff ff ff ff ff ff ff ff - ff ff ff ff ff ff ff ff
a5 a5 a5 a5 a5 a5 a5 a5 - a5 a5 a5 a5 a5 a5 a5 a5
a5 a5 a5 a5 a5 a5 a5 a5 - a5 a5 a5 a5 a5 a5 a5 a5
a5 a5 a5 a5 a5 a5 a5 a5 - a5 a5 a5 a5 a5 a5 a5 a5
a5 a5 a5 a5 a5 a5 a5 a5 - a5 a5 a5 a5 a5 a5 a5 a5
ff ff ff ff ff ff ff ff - ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff - ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff - ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff - ff ff ff ff ff ff ff ff

And now I can't get it to fail... so not sure what to do. I am concerned that there is a potential interaction with debugging and the flash but don't have any hard data to confirm.

henrikbrixandersen, 2020-09-09:
@dleach02 Would it be okay if I assign this to you instead? I am out of ideas on what the root cause might be...

nvlsianpu, 2020-09-09:
> I am concerned that there is a potential interaction with debugging and the flash but don't have any hard data to confirm.
Looks like potential a twr_ke18f SoC flash driver issue - might it be caused by a memory monitoring while a flash erase operation was in progress? Usually erase operation does erase entire flash page (is page size 2 KB?), so this looks strange, like page erase operation failure.
I don't know SoC used so can't really help here.

henrikbrixandersen, 2021-01-11:
@hakehuang How come you closed this issue?

hakehuang, 2021-01-11:
@henrikbrixandersen, I get below notice, are you still owrkin gon this?

> This issue has been marked as stale because it has been open (more than) 60 days with no activity. Remove the stale label or add a comment saying that you would like to have the label removed [...]

henrikbrixandersen, 2021-01-11:
> @henrikbrixandersen, I get below notice, are you still working on this?

No, I removed the stale label. The issue is still present.

hakehuang, 2021-01-11:
@henrikbrixandersen , at my latest teset I get below error
```
Running test suite test_flash_map
===================================================================
START - test_flash_area_erased_val
 PASS - test_flash_area_erased_val
===================================================================
START - test_flash_area_get_sectors
E: ***** BUS FAULT *****
E:   Instruction bus error
E:   NXP MPU error, port 3
E:     Mode: Supervisor, Instruction Address: 0x20000c68
E:     Type: Read, Master: 0, Regions: 0x8800
E: r0/a1:  0x40020000  r1/a2:  0x00070000  r2/a3:  0x00000fff
E: r3/a4:  0x20000c69 r12/ip:  0x20000ca8 r14/lr:  0x000037a1
E:  xpsr:  0x01000000
E: Faulting instruction address (r15/pc): 0x20000c68
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20000170 (unknown)
E: Halting system

```

MaureenHelm, 2021-02-09:
@dleach02 can you take another look at this?

closed 2021-04-26 by github-actions[bot]