zephyrproject-rtos/zephyr #27205: tests/kernel/timer/timer_api test fails on twr_ke18f

hakehuang, 2020-07-28:
**Describe the bug**
tests/kernel/timer/timer_api test fails on twr_ke18f
platform: 
twrke18f
Location:
              tests/kernel/timer/timer_api 

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=twr_ke18f ..
3. make flash
4. See error

**Expected behavior**
test pass

**Impact**
timer api

**Logs and console output**

```
    *** Booting Zephyr OS version 2.3.99  ***
    Running test suite timer_api
    ===================================================================
    START - test_time_conversions
     PASS - test_time_conversions
    ===================================================================
    START - test_timer_duration_period
     PASS - test_timer_duration_period
    ===================================================================
    START - test_timer_period_0
     PASS - test_timer_period_0
    ===================================================================
    START - test_timer_expirefn_null
     PASS - test_timer_expirefn_null
    ===================================================================
    START - test_timer_periodicity
     PASS - test_timer_periodicity
    ===================================================================
    START - test_timer_status_get
     PASS - test_timer_status_get
    ===================================================================
    START - test_timer_status_get_anytime
     PASS - test_timer_status_get_anytime
    ===================================================================
    START - test_timer_status_sync
     PASS - test_timer_status_sync
    ===================================================================
    START - test_timer_k_define
    
        Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:94: duration_expire: (interval >= 100) || (((10000 % 1000U) != 0) && (interval == 100 - 1)) is false
    
     FAIL - test_timer_k_define
    ===================================================================
    START - test_timer_user_data
     PASS - test_timer_user_data
    ===================================================================
    START - test_timer_remaining
     PASS - test_timer_remaining
    ===================================================================
    START - test_timeout_abs
     PASS - test_timeout_abs
    ===================================================================
    Test suite timer_api failed.
    ===================================================================
    PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA:  ab3e778f47d

hakehuang, 2020-07-28:
@MaureenHelm @dlech

dlech, 2020-07-28:
Not sure why I am being @ mentioned

MaureenHelm, 2020-07-28:
> Not sure why I am being @ mentioned

Wrong @. Should be @dleach02 instead.

hakehuang, 2020-07-29:
> Not sure why I am being @ mentioned

sorry, I did not checkl carefully.

henrikbrixandersen, 2020-07-29:
@hakehuang I have just tried this a couple of times on Zephyr master (5e2e7a0fcf6f9815080e5fb311808730ccf3cbeb). I cannot reproduce the error, you are seeing.

Can you provide more information?

hakehuang, 2020-07-31:
@henrikbrixandersen this issue is gone with build  on Zephyr master (5e2e7a0)

closed 2020-07-31 by hakehuang