zephyrproject-rtos/zephyr #27237: Out_of_tree example broken

a1ien, 2020-07-29:
**Describe the bug**
When try build out_of_tree_board I get `fatal error: nrfx.h: No such file or directory`

**To Reproduce**
Steps to reproduce the behavior:
```
$ tree -L 2      
.
└── zephyrproject
    ├── bootloader
    ├── modules
    ├── tools
    └── zephyr
$ cp -R zephyrproject/zephyr/samples/application_development/out_of_tree_board ./
$ cd out_of_tree_board 
$ cmake -B build -GNinja -DBOARD=nrf52840dk_nrf52840
$ cmake --build build
```
**Expected behavior**
Build without error

**Impact**
Can't build out of tree

**Logs and console output**
```
$ cmake -B build -GNinja -DBOARD=nrf52840dk_nrf52840 
Including boilerplate (Freestanding): /home/user/projects/zephyr/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/user/projects/zephyr/out_of_tree_board
-- Zephyr version: 2.3.99 (/home/user/projects/zephyr/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.7 (found suitable exact version "3.7.5") found components:  Interpreter 
-- Board: nrf52840dk_nrf52840
ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found toolchain: zephyr (/home/user/bin/zephyr-sdk-0.11.3)
-- Found west: /home/user/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /home/user/bin/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/user/projects/zephyr/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Generated zephyr.dts: /home/user/projects/zephyr/out_of_tree_board/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/user/projects/zephyr/out_of_tree_board/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/user/projects/zephyr/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/user/projects/zephyr/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration '/home/user/projects/zephyr/out_of_tree_board/prj.conf'
Configuration saved to '/home/user/projects/zephyr/out_of_tree_board/build/zephyr/.config'
Kconfig header saved to '/home/user/projects/zephyr/out_of_tree_board/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/bin/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/user/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/projects/zephyr/out_of_tree_board/build
$ cmake --build build
[1/126] Preparing syscall dependency handling

[7/126] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj 
/home/user/bin/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.3.0-1379-gcb5e45c88442 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I/home/user/projects/zephyr/zephyrproject/zephyr/kernel/include -I/home/user/projects/zephyr/zephyrproject/zephyr/arch/arm/include -I/home/user/projects/zephyr/zephyrproject/zephyr/include -Izephyr/include/generated -I/home/user/projects/zephyr/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52 -isystem /home/user/projects/zephyr/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/user/bin/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include -isystem /home/user/bin/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include-fixed -Os -imacros/home/user/projects/zephyr/out_of_tree_board/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros/home/user/projects/zephyr/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/user/projects/zephyr/out_of_tree_board=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/user/projects/zephyr/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/user/projects/zephyr/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj   -c /home/user/projects/zephyr/zephyrproject/zephyr/arch/arm/core/offsets/offsets.c
In file included from /home/user/projects/zephyr/zephyrproject/zephyr/include/arch/arm/aarch32/cortex_m/cmsis.h:17,
                 from /home/user/projects/zephyr/zephyrproject/zephyr/include/arch/arm/aarch32/cortex_m/mpu/arm_mpu_v7m.h:10,
                 from /home/user/projects/zephyr/zephyrproject/zephyr/include/arch/arm/aarch32/cortex_m/mpu/arm_mpu.h:13,
                 from /home/user/projects/zephyr/zephyrproject/zephyr/include/arch/arm/aarch32/arch.h:248,
                 from /home/user/projects/zephyr/zephyrproject/zephyr/include/arch/cpu.h:19,
                 from /home/user/projects/zephyr/zephyrproject/zephyr/include/kernel_includes.h:38,
                 from /home/user/projects/zephyr/zephyrproject/zephyr/include/kernel.h:17,
                 from /home/user/projects/zephyr/zephyrproject/zephyr/arch/arm/core/offsets/offsets_aarch32.c:28,
                 from /home/user/projects/zephyr/zephyrproject/zephyr/arch/arm/core/offsets/offsets.c:12:
/home/user/projects/zephyr/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52/soc.h:16:10: fatal error: nrfx.h: No such file or directory
   16 | #include <nrfx.h>
      |          ^~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain SDK
 - Commit cb5e45c88442

a1ien, 2020-07-29:
Broken after ef3c5e551676278ccc1f42ecb572838144c5e33a

MaureenHelm, 2020-08-25:
@tejlmand can you please take a look at this?

tejlmand, 2020-08-26:
> @tejlmand can you please take a look at this?

Thanks for pinging.
Fixed: https://github.com/zephyrproject-rtos/zephyr/pull/27797
Sorry for the delay, I haven't noticed this issue before.

PR #27797 (2020-08-26): cmake: zephyr_module.py working directory when listing modules
Fixes: #27237

This commit fixes an issue when `zephyr_module.py` was executed outside
a west workspace.
This would happen when build an out-of-tree (out-of-workspace)
application, in which case the current west workspace would be unknown.

This is now changed, so that execution of `zephyr_module.py` will be
done in the current Zephyr base.
This ensures that the west workspace that holds the current Zephyr will
be the same workspace used for west list, and hence solve fix the issue.
commit 6c294bc0 (2020-08-26): cmake: zephyr_module.py working directory when listing modules

Fixes: #27237

This commit fixes an issue when `zephyr_module.py` was executed outside
a west workspace.
This would happen when build an out-of-tree (out-of-workspace)
application, in which case the current west workspace would be unknown.

This is now changed, so that execution of `zephyr_module.py` will be
done in the current Zephyr base.
This ensures that the west workspace that holds the current Zephyr will
be the same workspace used for west list, and hence solve fix the issue.
closed 2020-08-26 by nashif