zephyrproject-rtos/zephyr #27239: samples/subsys/canbus/isotp/sample.subsys.canbus.isotp fails on FRDM-K64F

galak, 2020-07-29:
When trying to run this sample on a FRDM-K64F I get:

```
*** Booting Zephyr OS build zephyr-v2.3.0-667-g766bb84a95e2  ***
Start sending data
TX complete cb [-2]
Error while sending data to ID 384 [-2]
Receiving error [-14]
Got 0 bytes in total
Receiving erreor [-14]
[00:00:02.007,000] <err> isotp: Reception of next FC has timed out
[00:00:02.007,000] <err> isotp: Reception of next FC has timed out
TX complete cb [-2]
Error while sending data to ID 384 [-2]
Receiving error [-14]
Got 0 bytes in total
Receiving erreor [-14]
[00:00:04.012,000] <err> isotp: Reception of next FC has timed out
[00:00:04.013,000] <err> isotp: Reception of next FC has timed out
```

henrikbrixandersen, 2020-07-29:
I can reproduce the error here.

galak, 2020-07-29:
@alexanderwachter mentioned that the sample requires at least 2 boards connect together for this to function.

galak, 2020-08-04:
Need to add a fixture to the sample.yaml.

dleach02, 2020-08-13:
@galak, is this an actual bug or is this a test setup problem? The readme for this test says it needs two boards connected together. Also, where is the canbus connection on the frdm-k64f?

henrikbrixandersen, 2020-08-13:
@dleach02 The FRDM-K64 does not come with an onboard CAN transceiver. You have to connect and external CAN transceiver to PTB18 (CAN0_TX) and PTB19 (CAN0_RX).

galak, 2020-08-13:
We can close this.  The issue was fixed by adding a fixture to the sample since it requires 2 boards.  @henrikbrixandersen when you said you can reproduce the issue on the frdm-k64f board, was that with or without external transceiver and connection to another board?

henrikbrixandersen, 2020-08-13:
@galak Without external transceiver and no second board. I was under the impression that the test used loopback since no test fixture was mentioned.

henrikbrixandersen, 2020-08-17:
Fixed by #27499. Closing.

closed 2020-08-17 by henrikbrixandersen