zephyrproject-rtos/zephyr #27249: Is there any development plan for supporting RPL stack ？

icephyr, 2020-07-30:
I'm doing research on the RPL stack development currently , while I noticed that the RPL node demo was removed from zephyr since v1.14. I am not sure about the reason, maybe not fully supported yet ?

So I wonder if zephyr has plans to officially support fully implemented RPL stack and what is the progress now ? Hope anyone can share some information about this. Thanks a lot.

jukkar, 2020-07-30:
The reason the RPL code was removed in 1.14, was that there was nobody willing to maintain the RPL code. If there is proper commitment to support and maintain RPL code, then patches are welcome.

icephyr, 2020-07-31:
Well noted, thanks for the answer.

closed 2020-07-31 by icephyr