zephyrproject-rtos/zephyr #27266: samples: bluetooth: hci_spi: Invalid cmd_hdr and acl_hdr usage

CorentinDOROTHEE, 2020-07-30:
**Describe the bug**
cmd_hdr and acl_hdr local structs were read but never written.
Causing unpredictable errors in bt_tx_thread such as :
"Invalid HCI CMD packet length" in hci.c:2280
or Imprecise data bus errors on nrf52810.

I have pinpointed the issue to samples/bluetooth/hci_spi/src/main.c using SEGGER J-Link probe

**To Reproduce**
Steps to reproduce the behavior:
1. have an hci_spi hardware setup
2. Add these Kconfig option to hci_spi sample:
    CONFIG_BT_LOG_LEVEL_DBG=y
    CONFIG_BT_DEBUG_LOG=y
    CONFIG_BT_DEBUG_HCI_DRIVER=y
3. build and flash hci_spi for the controller
4. build and flash peripheral_hr for the host
5. See error on controller serial console

**Expected behavior**
cmd_hdr and acl_hdr local structs should be filled with the SPI data received from the host cpu

**Impact**
hci_spi is unusable and crash at random spots depending on the host application:
- Advertiser might not start
- Discovering bluetooth services won't work

**Environment (please complete the following information):**
 - OS: Linux x86_64 Kernel : 5.7.6-arch1-1
 - Toolchain : Zephyr SDK (0.11.3)
 - Commit : Tag v2.3.0

lochej, 2020-07-30:
@CorentinDOROTHEE Thank you very much for finding this bug !

In fact, cmd_hdr and acl_hdr are **not even initiailized with 0 values.**
So you get whatever is in the thread stack ram 🙃

This may be the reason why this bug was not detected before (due to its random occurance nature).

This problem is still present in master. 

The fix for this problem should be integrated in both **master and V2.3.0 release as backport.**

lochej, 2020-07-30:
A fix to this problem would be to either correctly fill the **cmd_hdr & acl_hdr** with spi data
using memcpy from rxmsg spi buffer.

Or simply remove the two structs and get the header length from rxmsg content directly. This could save a bit of the thread stack memory.

lochej, 2020-07-30:
Looks like this bug was introduced in this commit:

https://github.com/zephyrproject-rtos/zephyr/commit/4b622afbb33fa199078a2215c72a0b822ae6841b#diff-01757f00c05ae26127a503a0c08eb4f0

The author deleted the original memcpy calls filling up the headers.

PR #27292 (2020-08-04): samples: bluetooth: hci_spi: Fix cmd_hdr and acl_hdr usage
Fixes #27266.
cmd_hdr and acl_hdr local structs were read but never written.
Causing unpredictable errors in bt_tx_thread such as :
"Invalid HCI CMD packet length" in hci.c:2280
or Imprecise data bus errors on nrf52810.

cmd_hdr and acl_hdr are now filled with the received data
from SPI Master.
commit 37aca482 (2020-08-04): samples: hci_spi: Fix cmd_hdr and acl_hdr usage

Fixes #27266.
cmd_hdr and acl_hdr local structs were read but never written.
Causing unpredictable errors in bt_tx_thread such as :
"Invalid HCI CMD packet length" in hci.c:2280
or Imprecise data bus errors on nrf52810.

cmd_hdr and acl_hdr are now filled with the received data
from SPI Master.
closed 2020-08-04 by carlescufi