zephyrproject-rtos/zephyr #27268: usb: mcux RT1060 EVK - using on-chip memory fails to configure USB

rtalbott, 2020-07-30:
**Describe the bug**
USB fails to configure when using on-chip memory rather than the default external DDR RAM.

**To Reproduce**
Change the default DATA_LOCATION to DATA_OCRAM instead of DATA_SEMC.
(be sure to unplug the USB port between resets - I have seen it mask the issue if left plugged in.)

**Expected behavior**
USB should configure and work.

**Impact**
USB showstopper

**Logs and console output**
-- Using DATA_SEMC (normal and expected):
*** Booting Zephyr OS build zephyr-v2.3.0-1418-g68ef393d2551  ***
[00:00:00.005,000] <dbg> main.composite_pre_init: HID Device: dev 0x80002058
[00:00:00.005,000] <dbg> main.main: Starting application
[00:00:00.189,000] <dbg> main.status_cb: status 1 unhandled
[00:00:00.314,000] <dbg> main.status_cb: status 1 unhandled
[00:00:02.440,000] <dbg> main.send_report: Wrote 2 bytes with ret 0

-- Using DATA_OCRAM (bug):
*** Booting Zephyr OS build zephyr-v2.3.0-1418-g68ef393d2551  ***
[00:00:00.005,000] <dbg> main.composite_pre_init: HID Device: dev 0x20202058
[00:00:00.005,000] <dbg> main.main: Starting application
[00:00:00.006,000] <err> os: ***** BUS FAULT *****
[00:00:00.006,000] <err> os:   Precise data bus error
[00:00:00.006,000] <err> os:   BFAR Address: 0xdd416a2c
[00:00:00.006,000] <err> os: r0/a1:  0x20200600  r1/a2:  0xdd416a28  r2/a3:  0x00000001
[00:00:00.006,000] <err> os: r3/a4:  0x94cefdf7 r12/ip:  0x00000040 r14/lr:  0x60007d53
[00:00:00.006,000] <err> os:  xpsr:  0xa1002c00
[00:00:00.006,000] <err> os: Faulting instruction address (r15/pc): 0x60007d2a
[00:00:00.006,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.006,000] <err> os: Current thread: 0x20202320 (unknown)
[00:00:00.053,000] <err> os: Halting system

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA 68ef393d255152797d99a271487839d5a70139ab (July 29, 2020)

**Additional context**

I have located the general area where the fault is occurring:

usb_dc_mcux_ehci.c:  lines 203..
```
if (k_mem_pool_alloc(&ep_buf_pool, block, cfg->ep_mps, K_MSEC(10))) {
	LOG_ERR("Memory allocation time-out");
	return -ENOMEM;
}
```

rtalbott, 2020-07-30:
I found the issue to be due to how the endpoint memory pool is being created.

To work around the problem is to ignore the CONFIG_NOCACHE_MEMORY (set in a Kconfig):

usb_dc_mcux_ehci.c: 
```
45:  /* The max MPS is 1023 for FS, 1024 for HS. */
46:  // #if defined(CONFIG_NOCACHE_MEMORY) 
47:  // #define EP_BUF_NONCACHED
48:  // __nocache K_MEM_POOL_DEFINE(ep_buf_pool, 16, 1024, EP_BUF_NUMOF_BLOCKS, 4);
49:  // #else
50:  K_MEM_POOL_DEFINE(ep_buf_pool, 16, 1024, EP_BUF_NUMOF_BLOCKS, 4);
51:  // #endif

```
I'm not sure if this is the correct solution for all mcux, but it does address my particular issue using on-chip RAM for the RT series.

jfischer-phytec-iot, 2020-08-05:
@MaureenHelm Is it configuration issue? This does not seem right [select NOCACHE_MEMORY if HAS_MCUX_CACHE](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/usb/device/Kconfig#L88).

closed 2020-09-17 by MaureenHelm