zephyrproject-rtos/zephyr #27279: CMSIS RTOS v1 Signals Implementation Issue

anthonytex, 2020-07-31:
Hi Zephyr devs,

yesterday i've try zephyr for the first time.
My idea was to port a simple firmware that is written with CMSIS RTOS API v1 to Zephyr.
For this, i'm using zephyr's portability layer: `CONFIG_CMSIS_RTOS_V1=y`.

This firmware uses signals API to sync threads, but something is not working, here there is a simple program to test the issue:
```c
#include <kernel.h>
#include <cmsis_os.h>

void print_thread(void const *args)
{
    printk("Started Thread 1, waiting for signals\n");

    while (true)
    {
        osSignalWait(0x1, osWaitForever);
        printk("Signal Received\n");
    }
}

osThreadDef(print_thread, osPriorityNormal, 1, 256);

int main(void)
{
    osThreadId tid = osThreadCreate(osThread(print_thread), NULL);
    //Send signals every second
    while (true)
    {
        printk("Waiting\n");
        osDelay(1000);
        osSignalSet(tid, 0x1);
    }
}
```

Digging in, i've found 2 problems:

1. when the thread is created: https://github.com/zephyrproject-rtos/zephyr/blob/e90873f290e2351232358239152a0fbfde83ea7e/lib/cmsis_rtos_v1/cmsis_thread.c#L83 
the custom data that is used for signals handling in your CMSIS implementation is, for what i can see, set for the current running thread and not for the created new thread: so when you call https://github.com/zephyrproject-rtos/zephyr/blob/e90873f290e2351232358239152a0fbfde83ea7e/lib/cmsis_rtos_v1/cmsis_signal.c#L90 you are pointing to other thread data.
1. By assuming that the custom data was properly set,  then when the thread is created by calling https://github.com/zephyrproject-rtos/zephyr/blob/e90873f290e2351232358239152a0fbfde83ea7e/lib/cmsis_rtos_v1/cmsis_thread.c#L85 the custom data is reset by this https://github.com/zephyrproject-rtos/zephyr/blob/e90873f290e2351232358239152a0fbfde83ea7e/kernel/thread.c#L568

So signals handling for me is not working, a simple hack fixes the problem (only for demostration, sorry but i've not a deep knowledge of zephyr so feel free to fix this in a better way of course):
```c
cm_thread = thread_def->cm_thread;
	atomic_dec((atomic_t *)&thread_def->instances);
	stk_ptr = thread_def->stack_mem;
	prio = cmsis_to_zephyr_priority(thread_def->tpriority);
	//k_thread_custom_data_set((void *)thread_def);

	tid = k_thread_create(&cm_thread[thread_def->instances],
			stk_ptr[thread_def->instances], stacksz,
			(k_thread_entry_t)zephyr_thread_wrapper,
			(void *)arg, NULL, thread_def->pthread,
			prio, 0, K_NO_WAIT);
       tid->custom_data = (void *)thread_def;

```

Also your test suit does not cover this case, in fact the *current thread* is waiting to be signaled and not the *newly created* like in my example:
https://github.com/zephyrproject-rtos/zephyr/blob/e90873f290e2351232358239152a0fbfde83ea7e/tests/portability/cmsis_rtos_v1/src/signal.c#L72-L90

Thanks

carlescufi, 2020-07-31:
@anthonytex thanks for the extensive analysis. Perhaps you'd like to send a Pull Request fixing the behavior?

anthonytex, 2020-07-31:
Hi @carlescufi , i've just open a PR but for another issue. In these days i'm going to find some time to send a PR to fix this.
Thanks

closed 2020-12-07 by github-actions[bot]