zephyrproject-rtos/zephyr #27282: Drivers in app folder

thomasoggiersoprod, 2020-07-31:
Hi,
I want to add new driver in my app folder.
When I compile and run it, I can get the node in device tree, but the function device_get_binding return NULL.
I also add a print in the init function of my driver, and this one is never called.
If I place this driver into zephyr's folder, the driver is found.
So if someone can help me

carlescufi, 2020-07-31:
Have you made sure to follow the practices shown in the sample?
https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/application_development/out_of_tree_driver

thomasoggiersoprod, 2020-07-31:
Thanks, I've already saw this sample. but it don't help me.

tejlmand, 2020-08-06:
It would be nice if you could put some code snippets of what you try to do, but based on your description, I assume that you are trying to use:
```
zephyr_library()
zephyr_library_sources(<some_driver.c>)
```

in a file like `app/drivers/CMakeLists.txt`

So that `app/CMakeLists.txt` looks like:
```
find_package(Zephyr)

add_subdirectory(drivers)
```

Now, the problem with such approach, is that the linking command needs to know all Zephyr libraries in order to correctly place them in the `--whole-archive` section:
https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/linker/ld/target.cmake#L92-L94

All of that is taken care of as part of `find_package(Zephyr)`, but if a user calls `zephyr_library()` AFTER `find_package(Zephyr)`, then that library will not be part of this.

Therefore the correct approach is to create the driver in a Zephyr module.
You can use the example given by @carlescufi .

Also see this comment: https://github.com/zephyrproject-rtos/zephyr/issues/27181#issuecomment-669824041

carlescufi, 2020-08-14:
Closing since @tejlmand already provided an answer

closed 2020-08-14 by carlescufi