zephyrproject-rtos/zephyr #27288: linker relocation feature fails for out of tree projects

markrowe, 2020-07-31:
I tried using the relocation feature per the Zephyr documentation:

added these to prj.conf
CONFIG_CODE_DATA_RELOCATION=y
CONFIG_HAVE_CUSTOM_LINKER_SCRIPT=y
CONFIG_CUSTOM_LINKER_SCRIPT="linker_arm_sram2.ld"

First tried the example in the sample/application_development area ... which works.
I applied it to our out of tree board but the python script that generates the location script fails.  It looks to me like the script doesn't like out of tree boards because it chokes on the  CMakeList.txt entry "**zephyr_code_relocate(${UI}/graphics.c SRAM2_BSS)" saying it expects 2 parameters and only got one.
Attached the linker file.
[linker_arm_sram2.zip](https://github.com/zephyrproject-rtos/zephyr/files/5006776/linker_arm_sram2.zip)

**Expected behavior**
I expected the python script to complete normally.

**Impact**
I was able to get around it by just using a local link directive file but it is a hack and may have issues going forward.

**Logs and console output**

**Environment (please complete the following information):**
 - Linux
 - Zephyr SDK 0.11.1
 - Zephyr 2.3.0  (really Nordic's ncs-1.3.0)

**Additional context**
Add any other context about the problem here.

MaureenHelm, 2020-08-04:
Can you reproduce this issue in upstream zephyr (not nordic ncs)?

markrowe, 2020-08-10:
Good question.  That will take some work.

tejlmand, 2020-08-10:
Would also be nice with a little more info on you structure.
Is you out-of-tree code being included as a Zephyr module ?
https://docs.zephyrproject.org/latest/guides/modules.html

tejlmand, 2020-09-08:
@markrowe any success in reproducing this in plain Zephyr ?

Is your project a `Zephyr workspace application` or a `Zephyr freestanding application` ?
See the differences here:
https://docs.zephyrproject.org/latest/guides/zephyr_cmake_package.html#zephyr-workspace-application
https://docs.zephyrproject.org/latest/guides/zephyr_cmake_package.html#zephyr-freestanding-application

tejlmand, 2020-09-09:
Did some further investigation on this:

Moved board out of Zephyr repo, and into a workspace repo, run cmake+ninja:
```
$ mkdir -p ../oot_project/boards/arm/; mv boards/arm/nrf52840dk_nrf52840/ ../oot_project/boards/arm/
$ rm -rf build; cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -DBOARD_ROOT=/projects/github/ncs/oot_project/ -Bbuild -DCONFIG_CODE_DATA_RELOCATION=y -DCONFIG_HAVE_CUSTOM_LINKER_SCRIPT=y -DCONFIG_CUSTOM_LINKER_SCRIPT=\"linker_arm_sram2.ld\" samples/application_development/code_relocation/
Including boilerplate (Zephyr repository): /projects/github/ncs/zephyr/cmake/app/boilerplate.cmake
-- Application: /projects/github/ncs/zephyr/samples/application_development/code_relocation
...
-- Found BOARD.dts: /projects/github/ncs/oot_project/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
...
-- Generating done
-- Build files have been written to: /projects/github/ncs/zephyr/build
$ ninja -Cbuild 
ninja: Entering directory `build'
[1/139] Preparing syscall dependency handling

[134/139] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           SRAM2:         524 B       128 KB      0.40%
           FLASH:       16406 B         1 MB      1.56%
            SRAM:        5686 B       256 KB      2.17%
        IDT_LIST:          56 B         2 KB      2.73%
[139/139] Linking C executable zephyr/zephyr.elf
```
works without errors

Moved the sample out of Zephyr repo, updated `./../../kernel/sem.c` to absolute path `${ZEPHYR_BASE}/kernel/sem.c`.
Code is still in workspace, did cmake+ninja.

This reveals a different error than described in this PR, that abs path is not working.
```
[129/139] Generating code_relocation.c, include/generated/linker_relocate.ld
/projects/github/ncs/zephyr/scripts/gen_relocate_app.py:402: UserWarning: File: /projects/github/ncs/oot_project/samples/application_development/code_relocation//projects/github/ncs/zephyr/kernel/sem.c Not found
  warnings.warn("File: "+file_name+" Not found")
```

This is fixed now, and build succeeds.
```
$ mkdir -p ../oot_project/samples/application_development/; mv ./samples/application_development/code_relocation/ ../oot_project/samples/application_development/
$ rm -rf build; cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -DBOARD_ROOT=/projects/github/ncs/oot_project/ -Bbuild -DCONFIG_CODE_DATA_RELOCATION=y -DCONFIG_HAVE_CUSTOM_LINKER_SCRIPT=y -DCONFIG_CUSTOM_LINKER_SCRIPT=\"linker_arm_sram2.ld\" ../oot_project/samples/application_development/code_relocation/
Including boilerplate (Zephyr workspace): /projects/github/ncs/zephyr/cmake/app/boilerplate.cmake
-- Application: /projects/github/ncs/oot_project/samples/application_development/code_relocation
-- Zephyr version: 2.4.0-rc1 (/projects/github/ncs/zephyr)
...
-- Generating done
-- Build files have been written to: /projects/github/ncs/zephyr/build
$ ninja -Cbuild 
ninja: Entering directory `build'
[1/139] Preparing syscall dependency handling

[134/139] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           SRAM2:         524 B       128 KB      0.40%
           FLASH:       16406 B         1 MB      1.56%
            SRAM:        5686 B       256 KB      2.17%
        IDT_LIST:          56 B         2 KB      2.73%
[139/139] Linking C executable zephyr/zephyr.elf
```

Moved everything to `/tmp` as Zephyr freestanding application.
```
$ mv ../oot_project /tmp/
$ rm -rf build; cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -DBOARD_ROOT=/tmp/oot_project/ -Bbuild -DCONFIG_CODE_DATA_RELOCATION=y -DCONFIG_HAVE_CUSTOM_LINKER_SCRIPT=y -DCONFIG_CUSTOM_LINKER_SCRIPT=\"linker_arm_sram2.ld\" /tmp/oot_project/samples/application_development/code_relocation/
Including boilerplate (Freestanding): /projects/github/ncs/zephyr/cmake/app/boilerplate.cmake
-- Application: /tmp/oot_project/samples/application_development/code_relocation
-- Zephyr version: 2.4.0-rc1 (/projects/github/ncs/zephyr)
...
-- Generating done
-- Build files have been written to: /projects/github/ncs/zephyr/build
$ ninja -Cbuild 
ninja: Entering directory `build'
[1/139] Preparing syscall dependency handling

[134/139] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           SRAM2:         524 B       128 KB      0.40%
           FLASH:       16406 B         1 MB      1.56%
            SRAM:        5686 B       256 KB      2.17%
        IDT_LIST:          56 B         2 KB      2.73%
[139/139] Linking C executable zephyr/zephyr.elf
```
which works with absolute path patch.

PR #28208 (2020-09-10): cmake: support absolute path in zephyr_code_relocate() function
Fixes: #27288

The Zephyr CMake function `zephyr_code_relocate()` would treat any file
argument as relative to `CMAKE_CURRENT_SOURCE_DIR` even if an absolute
path is provided.

A file with absolute path would thus become
`${CMAKE_CURRENT_SOURCE_DIR}/<absolute-path>/<file>` which leads to a
non existing file.

This is fixed by checking if path argument is a relative path, and only
use `CMAKE_CURRENT_SOURCE_DIR` in such case.
commit eb2a7054 (2020-09-10): cmake: support absolute path in zephyr_code_relocate() function

Fixes: #27288

The Zephyr CMake function `zephyr_code_relocate()` would treat any file
argument as relative to `CMAKE_CURRENT_SOURCE_DIR` even if an absolute
path is provided.

A file with absolute path would thus become
`${CMAKE_CURRENT_SOURCE_DIR}/<absolute-path>/<file>` which leads to a
non existing file.

This is fixed by checking if path argument is a relative path, and only
use `CMAKE_CURRENT_SOURCE_DIR` in such case.
closed 2020-09-10 by MaureenHelm