zephyrproject-rtos/zephyr #27293: Test nrf52840dk_nrf52840 tests/net/socket/net_mgmt/net.socket.mgmt fails on master

hubertmis, 2020-07-31:
**Describe the bug**
I've created PR #27284 . Buildkite reported failure. I've reproduced the issue locally. However, it does not look like a failure caused by my changes in the pull request. I've reproduced the same steps on the master branch and the test has failed.

**To Reproduce**
Steps to reproduce the behavior:
1. cd tests/net/socket/udp
2. west build -b nrf52840dk_nrf52840
3. See error

**Expected behavior**
Passing test on master branch

**Impact**
I cannot merge my PR, because buildkite blocks merging if it reports failures. I'm blocked with progressing on the feature I'm implementing.

**Logs and console output**
```
[179/185] Building C object zephyr/CMakeFiles/output_lib.dir/kobject_hash.c.obj
FAILED: zephyr/CMakeFiles/output_lib.dir/kobject_hash.c.obj 
ccache /opt/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc -DKERNEL -DNRF52840_XXAA -DUSE_PARTITION_MANAGER=0 -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I../../../../../include -Izephyr/include/generated -I../../../../../soc/arm/nordic_nrf/nrf52 -I../../../../../lib/libc/newlib/include -I../../../../../subsys/net/lib/sockets/. -I../../../../../subsys/testsuite/include -I../../../../../subsys/testsuite/ztest/include -I../../../../../subsys/net/l2 -I/home/humi/git/ncs/nrf/include -I/home/humi/git/ncs/modules/hal/cmsis/CMSIS/Core/Include -I/home/humi/git/ncs/modules/hal/nordic/nrfx -I/home/humi/git/ncs/modules/hal/nordic/nrfx/drivers/include -I/home/humi/git/ncs/modules/hal/nordic/nrfx/mdk -I/home/humi/git/ncs/modules/hal/nordic/. -I/home/humi/git/ncs/modules/debug/segger/rtt -Os -imacros/home/humi/git/ncs/zephyr/tests/net/socket/udp/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros/home/humi/git/ncs/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/humi/git/ncs/zephyr/tests/net/socket/udp=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/humi/git/ncs/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/humi/git/ncs=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs -fno-function-sections -fno-data-sections -MD -MT zephyr/CMakeFiles/output_lib.dir/kobject_hash.c.obj -MF zephyr/CMakeFiles/output_lib.dir/kobject_hash.c.obj.d -o zephyr/CMakeFiles/output_lib.dir/kobject_hash.c.obj   -c zephyr/kobject_hash.c
kobject_hash.gperf:62:38: error: 'K_OBJ_DRIVER_Z_THREAD_ST' undeclared here (not in a function)
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: gcc-arm-none-eabi-9-2019-q4-major 
 - Commit SHA: b83e4224e006fccb20e624a5f6e5170b19e6ea1d

**Additional context**
None

carlescufi, 2020-08-04:
This is indeed a bug in Zephyr master too:

```
FAILED: zephyr/CMakeFiles/output_lib.dir/kobject_hash.c.obj
ccache /home/carles/bin/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.3.0-1578-g7d920ba39b41 -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I../include -Izephyr/include/generated -I../soc/arm/nordic_nrf/nrf52 -I../lib/libc/newlib/include -I../subsys/net/lib/sockets/. -I../subsys/testsuite/include -I../subsys/testsuite/ztest/include -I../subsys/net/l2 -I/home/carles/src/zephyr/modules/hal/cmsis/CMSIS/Core/Include -I/home/carles/src/zephyr/modules/hal/nordic/nrfx -I/home/carles/src/zephyr/modules/hal/nordic/nrfx/drivers/include -I/home/carles/src/zephyr/modules/hal/nordic/nrfx/mdk -I/home/carles/src/zephyr/modules/hal/nordic/. -I/home/carles/src/zephyr/modules/debug/segger/rtt -Os -imacros/home/carles/src/zephyr/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros/home/carles/src/zephyr/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/carles/src/zephyr/zephyr/tests/net/socket/udp=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/carles/src/zephyr/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/carles/src/zephyr=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs -fno-function-sections -fno-data-sections -MD -MT zephyr/CMakeFiles/output_lib.dir/kobject_hash.c.obj -MF zephyr/CMakeFiles/output_lib.dir/kobject_hash.c.obj.d -o zephyr/CMakeFiles/output_lib.dir/kobject_hash.c.obj   -c zephyr/kobject_hash.c
kobject_hash.gperf:54:38: error: 'K_OBJ_DRIVER_Z_THREAD_ST' undeclared here (not in a function)
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/carles/bin/cmake/cmake-3.13.1-Linux-x86_64/bin/cmake --build /home/carles/src/zephyr/zephyr/build
```

@jukkar or @andrewboie any ideas?

joerchan, 2020-08-05:
Looks like it is coming from gen_kobject_list.py
There is an invalid element in the output kobject_hash.gperf
```
"\x68\x91\x00\x20", {}, K_OBJ_DRIVER_GPIO, 0 | K_OBJ_FLAG_DRIVER, { .unused = 0 }
"\x78\x91\x00\x20", {}, K_OBJ_DRIVER_Z_THREAD_ST, 0 | K_OBJ_FLAG_DRIVER, { .unused = 0 }
"\xb8\x91\x00\x20", {}, K_OBJ_MEM_SLAB, 0 | K_OBJ_FLAG_INITIALIZED, { .unused = 0 }
```
Z_THREAD_ST, looks like it has been truncated from `Z_THREAD_STACK_BUFFER`
```
strings build/zephyr/zephyr_prebuilt.elf| grep Z_THREAD_ST
Z_THREAD_STACK_BUFFER
```
And the address of it looks invalid:
```
arm-none-eabi-addr2line -e build/zephyr/zephyr_prebuilt.elf 0x20009178        
arm_mpu_regions.c:?
```

PerMac, 2020-08-06:
I think it might be some dependencies issue. In the daily runs on our CI I see passes for these test. Locally I got fail, but when I follow the reqs. setting from the CI plan I managed to get passes on master. I can try to find where is the difference

joerchan, 2020-08-06:
> This is indeed a bug in Zephyr master too:

@carlescufi Can you doublecheck this? Both me and @PerMac got the same result, failing on this PR, working on master.

hubertmis, 2020-08-06:
It looks that the test fails on the commit I've used as a base for my PR, but it is fixed now on the master.

joerchan, 2020-08-06:
> It looks that the test fails on the commit I've used as a base for my PR, but it is fixed now on the master.

@hubertmis If that is the case you should rebase your PR then.

@andrewboie Is this a known issue that has already been fixed, or are we just getting lucky here?

hubertmis, 2020-08-06:
@joerchan , thanks. Just did and waiting for buildkite report to close this issue.

hubertmis, 2020-08-06:
I'm closing the issue. It looks that reported problem has been fixed.

closed 2020-08-06 by hubertmis