zephyrproject-rtos/zephyr #27299: espi: xec: Whenever eSPI host indicates we are entering DnX mode, notification doesn't reach eSPI slave

albertofloyd, 2020-07-31:
**Describe the bug**
eSPI host sends multiple notification to eSPI slaves, platform reset, channel reset.
One of these notifications is entering DnX (Download and execute) mode, currently this notification is not propagated from the driver to app layer.

**To Reproduce**
1. cd samples/drivers/espi 
2.  west build -c -b mec1501modular_assy6885

3. No notification of DnX warn seen when forcing DnX mode using volume keys.

**Expected behavior**
Callback is received in eSPI driver sample indicated VW DnX WARN has arrived.

carlescufi, 2020-09-08:
@albertofloyd is this still an issue? If not please close it.

albertofloyd, 2020-09-09:
Still an issue, fix under testing.

closed 2020-11-23 by github-actions[bot]