zephyrproject-rtos/zephyr #27303: RFC: downgrade i2c eeprom_slave driver to sample

pabigot, 2020-07-31:
We have a [driver accessed under `CONFIG_I2C_EEPROM_SLAVE`](https://docs.zephyrproject.org/latest/reference/peripherals/i2c_eeprom_slave.html)  that implements the I2C follower (slave) driver API.  This device isn't an actual hardware driver: it's a virtual EEPROM that stores data in an instance-specific RAM buffer, with the data exposed on an I2C bus as a I2C follower device that can be controlled by another device acting as a leader (master) on that same bus.

As such it's a reasonable example of how to write an I2C follower driver, but it's not clear that it has a real use in applications: the sole in-tree reference is in the `i2c_slave_api` test.

Can anybody make a case for having this "device" available to applications?  If not, I propose to move it into the test source directory.

pabigot, 2020-07-31:
@henrikbrixandersen @superna9999 your input requested, please.

henrikbrixandersen, 2020-07-31:
@pabigot That is a good idea. I came across the that driver when I introduced the EEPROM API, and wondered what the use-case was.

carlescufi, 2020-08-04:
API meeting 4th August 2020:

- No objections in the meeting
- This unmaintained driver will be removed and a note will be added to the release notes, along with information on how to use the sample instead.

pabigot, 2020-08-04:
Mail sent 2020-08-04

PR #27387 (2020-08-13): cleanup of i2c_slave driver and test implementation and documentation
Closes #27303 by moving the simulated EEPROM I2C follower device into the test that uses it.

Closes #26805 by documenting how the test is intended to be used.

See individual commit messages for more details.

This commit includes material and concepts that derive from changes originally proposed by @sjg20 in  #26003.
closed 2020-08-13 by carlescufi