zephyrproject-rtos/zephyr #27319: [Coverity CID :211523] Bad bit shift operation in arch/arc/core/mpu/arc_mpu_v2_internal.h

zephyrbot, 2020-08-01:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/7d90812f265c0d23bad904434cef9c1616ba08ad/arch/arc/core/mpu/arc_mpu_v2_internal.h#L40

Category: Integer handling issues
Function: `_region_init`
Component: Architectures
CID: [211523](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=211523)

Details:

```
34         index = index * 2U;
35    
36         if (bits < ARC_FEATURE_MPU_ALIGNMENT_BITS) {
37             bits = ARC_FEATURE_MPU_ALIGNMENT_BITS;
38         }
39    
>>>     CID 211523:  Integer handling issues  (BAD_SHIFT)
>>>     In expression "1 << bits", left shifting by more than 31 bits has undefined behavior.  The shift amount, "bits", is 255.
40         if ((1 << bits) < size) {
41             bits++;
42         }
43    
44         if (size > 0) {
45             region_attr &= ~(AUX_MPU_RDP_SIZE_MASK);
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

dleach02, 2020-08-15:
The bits value is assigned a from find_msb_set() which can be a value of 0 to 32. But if it is 0 then the continuation logic for the assignment also subtracts 1 so now bits would equal 0xFF which will result in the undefined behavior coverity is complaining about. This code is in _region_init() where 'size' is a parameter and there are numerous instances where this function is called with size = 0.

``` 
uint8_t bits = find_msb_set(size) - 1;
```

ruuddw, 2020-08-17:
Thanks David, my analysis was the same, just didn't get to submitting a fix. Solution will be to move the find msb and shift logic to inside the later block guarded with 'if size>0'.
Btw, it could be considered a false positive as well, as the possibly incorrectly shifted bits are not used if size==0.
Still, moving the find msb and shifting to inside the loop is cleaner and (depending on compiler optimizations) probably slightly more efficient.

dleach02, 2020-08-17:
Good point on the false positive. I didn't think about that. But I think it is cleaner to do what you said and move the bit manipulation stuff into the if (size>0) block. Can you submit a PR for this so we can close this issue?

abrodkin, 2020-08-18:
@IRISZZW could you please prepare a PR for this one?

ruuddw, 2020-08-18:
> @IRISZZW could you please prepare a PR for this one?

I started on it already...

PR #27652 (2020-08-27): arc: Fix for undefined shift behavior (CID 211523)
find_msb_set() can return 0, resulting in a undefined shift of 255 bits.
Fixes #27319
commit 80194192 (2020-08-27): arc: Fix for undefined shift behavior (CID 211523)

find_msb_set() can return 0, resulting in a undefined shift of 255 bits.
Fixes #27319
closed 2020-08-27 by nashif