zephyrproject-rtos/zephyr #27330: include in prj.conf

Delsian, 2020-08-02:
**Is your feature request related to a problem? Please describe.**
Different subprojects in single superproject has a lot of common setting. On every superproject's change I need to replace the same values in all subprojects.

**Describe the solution you'd like**
Implement "#include <common.conf>" directive inside prj.conf

jukkar, 2020-08-03:
You can use overlays instead. You can do this by giving `-DOVERLAY_CONFIG=overlay-foobar.conf` in cmake. Then you could place the common settings into the overlay file and use it in your subprojects.

Delsian, 2020-08-03:
Ok, I see the right solution, using CONF_FILE directive in CMakeList.txt:

`set(CONF_FILE
  prj.conf
  log.conf
)`

Sorry for the wrong request

closed 2020-08-03 by Delsian