zephyrproject-rtos/zephyr #27331: Fails to upload over BLE using Zephyr with SMP Server Sample

caco3, 2020-08-02:
Hi all

I am trying to get the "SMP Server Sample" (https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html) working.

I have a nrf52840dongle_nrf52840 and successfully installed MCUboot (option #2) and the "SMP Server Sample" on it.
I also have a laptop with a built-in BLE radio.
I successfully can test the connection and even list the image:
```
> mcumgr --conntype ble --connstring ctlr_name= peer_name='Zephyr' echo hello 
hello
```
```
> sudo mcumgr --conntype ble --connstring ctlr_name=peer_name='Zephyr' image list
 Images:
  image=0 slot=0
    version: 0.0.0
    bootable: true
    flags: active confirmed
    hash: 37a4a9403fbd0a0336152646947ccce5faf6674e3a29b4220eb2b97a2f613b9d
Split status: N/A (0)
```

How ever when I try to upload a new image, I run into an error:
```
> sudo mcumgr --conntype ble --connstring peer_name='Zephyr' image upload smp_svr.signed.bin -t 100
 0 B / 182.61 KiB [--------------------------------------------------------------------------------------------------]   0.00%
 panic: send on closed channel
 goroutine 493 [running]:
  mynewt.apache.org/newtmgr/nmxact/nmp.(*Listener).AfterTimeout.func1()
    /home/ubuntu/go/src/mynewt.apache.org/newtmgr/nmxact/nmp/dispatch.go:50 +0x64
  created by time.goFunc
    /usr/lib/go-1.13/src/time/sleep.go:168 +0x44
```

Any Idea how this can get fixed?

carlescufi, 2020-08-03:
@caco3 this looks like an issue in the mcumgr side. Since this is maintained separately I suggest you contact the maintainers directly: https://github.com/apache/mynewt-mcumgr-cli

They have a Slack channel over at mynewt.slack.com. Also cc @utzig.

closed 2020-08-03 by carlescufi

utzig, 2020-08-03:
> Any Idea how this can get fixed?

Try installing `newtmgr` instead: `go get github.com/apache/mynewt-newtmgr/newtmgr`. Everywhere you see usage of `mcumgr` in the docs, use `newtmgr` instead. If this ends up with the same results, open an issue here: https://github.com/apache/mynewt-mcumgr-cli, and if it works just let me know that I will update mcumgr.

caco3, 2020-08-03:
Thanks @utzig 

It did not work either with newtmgr, so I opened an issue there: https://github.com/apache/mynewt-mcumgr-cli/issues/15