zephyrproject-rtos/zephyr #27338: Bluetooth: host: GATT service request is not able to trigger the authentication procedure while in SC only mode

ryanjh, 2020-08-03:
**Describe the bug**
While the Zephyr host is in SC only mode, it is not able to trigger the authentication procedure when accessing an GATT characteristic requiring READ_AUTHEN permission. In the case of insufficient authentication, the security level will be escalated gradually (att.c#L1976). As stated in the SPEC, if a device is in SC only mode, it shall only use security level 1 or level 4.

https://github.com/zephyrproject-rtos/zephyr/blame/master/subsys/bluetooth/host/att.c#L1976

Here's a relevant section from the Bluetooth Core Specification v5.2, Vol. 3, Part C, section 10.2.4:

>A device may be in a Secure Connections Only mode. When in Secure Connections Only mode only security mode 1 level 4 shall be used except for services that only require security mode 1 level 1.
The device shall only accept new outgoing and incoming service level connections for services that require Security Mode 1, Level 4 when the remote device supports LE Secure Connections and authenticated pairing is used.

**Expected behavior**
If a device is in SC only mode, and it gets a ATT response with BT_ATT_ERR_AUTHENTICATION, it requires to set sec to BT_SECURITY_L4 as a parameter of bt_conn_set_security().

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/conn.c#L1135

**Impact**
The Zephyr doesn't compliant the test procedure of BLE host qualification GAP/SEC/SEM/BV-23-C (GAP.TS.p38). 

**Logs and console output**

[image]

**Environment (please complete the following information):**

 - Commit SHA or Version used: Current master ([26216d1](https://github.com/zephyrproject-rtos/zephyr/commit/26216d1))

joerchan, 2020-08-03:
Same for L2CAP.

Vudentz, 2020-08-05:
Isn't the following code suppose to bump the security to SC automatically:

```
	if ((rsp->auth_req & BT_SMP_AUTH_SC) &&
	    (req->auth_req & BT_SMP_AUTH_SC)) {
		atomic_set_bit(smp->flags, SMP_FLAG_SC);

		rsp->init_key_dist &= RECV_KEYS_SC;
		rsp->resp_key_dist &= SEND_KEYS_SC;
	}
```

So I suppose we do actually end up with the right authentication.

joerchan, 2020-08-06:
@Vudentz No, that is LE Secure Connections pairing. It will be used if both peers support it, that is true, but it doesn't impact the security level. You can use LE Secure Connections pairing and still only be at L2.

In Secure Connections Only Mode the requirement is L4, i.e LE Secure Connections pairing, authentication (MITM) and 16-bit strength encryption key. When this is enabled calling bt_conn_set_security is only allowed to give L4 as input.

```
	if (IS_ENABLED(CONFIG_BT_SMP_SC_ONLY) &&
	    sec < BT_SECURITY_L4) {
		return -EOPNOTSUPP;
	}
```

PR #27341 (2020-08-24): Bluetooth: host: Elevate security to L4 when SC only is enabled
Elevate connections always to security mode 1 level 4 when
Secure Connections Only Mode has been enabled in the Security Manager.
    
Elevate connections always to security mode 1 level 3 when
Legacy pairing with OOB only has been enabled in the Security Manager.

Fixes: #27338
commit c7ad6615 (2020-08-24): Bluetooth: host: Elevate security to L4 when SC only is enabled

Elevate connections always to security mode 1 level 4 when
Secure Connections Only Mode has been enabled in the Security Manager.

Elevate connections always to security mode 1 level 3 when
Legacy pairing with OOB only has been enabled in the Security Manager.

Fixes: #27338
closed 2020-08-24 by carlescufi