zephyrproject-rtos/zephyr #27348: When using CONFIG_NVS it triggers BUS FAULT during startup on "nucleo_wb55rg" board

wojciiii, 2020-08-03:
**Describe the bug**
I would like to use flash to store Bluetooth LE settings when a LE bonding is complete as otherwise the device forgets that it has been bonded with another device.

In order to do this I enable:

"CONFIG_BT_SETTINGS=y
CONFIG_FLASH=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_FLASH_MAP=y
CONFIG_NVS=y
CONFIG_SETTINGS=y"

in the config.

The following is shown on the serial console:
*** Booting Zephyr OS build zephyr-v2.3.0-897-g7da5f15e1577  ***
[00:00:00.130,000] <dbg> os.k_sched_unlock: scheduler unlocked (0x20000ec8:0)
[00:00:00.131,000] <dbg> os.z_impl_k_mutex_lock: 0x20000ec8 took mutex 0x20001084, count: 1, orig prio: 0
[00:00:00.131,000] <err> os: ***** BUS FAULT *****
[00:00:00.131,000] <err> os:   Precise data bus error
[00:00:00.131,000] <err> os:   BFAR Address: 0x80c8ff8
[00:00:00.131,000] <err> os: r0/a1:  0x200027b8  r1/a2:  0x080c8ff8  r2/a3:  0x00000008
[00:00:00.131,000] <err> os: r3/a4:  0x200027b8 r12/ip:  0x00000000 r14/lr:  0x08002a23
[00:00:00.131,000] <err> os:  xpsr:  0x81000200
[00:00:00.131,000] <err> os: Faulting instruction address (r15/pc): 0x0800f7ec
[00:00:00.131,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.131,000] <err> os: Current thread: 0x20000ec8 (unknown)
[00:00:00.210,000] <err> os: Halting system

**To Reproduce**
Steps to reproduce the behaviour:

Build samples/bluetooth/peripheral_hids for "nucleo_wb55rg" board. 
Flash board.

(Removing CONFIG_NVS starts the application but fails to write settings to flash (I wanted to test that CONFIG_NVS options leads to corruption/crash).)

**Expected behavior**
I'm not expecting to see a BUS FAULT but settings being written to flash.

**Impact**
The Zephyr Bluetooth samples are working without saving any information to flash, but for anything serious on this boards I need to be able to write settings to flash.

**Logs and console output**

The crash output above describes the problem.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 LTS
 - Toolchain: zephyr-sdk-0.11.3
 - Commit SHA: 7da5f15e1577755b4d01dc05b8fac86f2da8cba8

**Additional context**

Is this a known problem on this board that flash writes doesn't work?

How can I debug startup of Zephyr - please point me to some docs?  When I do "west debug" gdb attaches at the point where the target already crashed.

Thanks. :)

pabigot, 2020-08-04:
Using PoP!_OS 20.04 (nearly Ubuntu) and zephyr-sdk-0.11.4 I'm unable to reproduce this on current master:
```
*** Booting Zephyr OS build zephyr-v2.3.0-1579-gc8bc3753aaf9  ***
[00:00:00.137,000] <inf> fs_nvs: 3 Sectors of 4096 bytes
[00:00:00.137,000] <inf> fs_nvs: alloc wra: 0, ff0
[00:00:00.137,000] <inf> fs_nvs: data wra: 0, 0
Bluetooth initialized
Advertising successfully started
[00:00:00.353,000] <inf> bt_hci_core: No ID address. App must call settings_loa)
[00:00:00.355,000] <inf> bt_hci_core: Identity: 80:e1:26:07:f0:6a (public)
[00:00:00.355,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0026, 0
[00:00:00.355,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x2126
```
or on the referenced commit:
```
*** Booting Zephyr OS build zephyr-v2.3.0-897-g7da5f15e1577  ***
[00:00:00.137,000] <inf> fs_nvs: 3 Sectors of 4096 bytes
[00:00:00.137,000] <inf> fs_nvs: alloc wra: 0, fd8
[00:00:00.137,000] <inf> fs_nvs: data wra: 0, 20
Bluetooth initialized
Advertising successfully started
[00:00:00.352,000] <inf> bt_hci_core: No ID address. App must call settings_loa)
[00:00:00.355,000] <inf> bt_hci_core: Identity: 80:e1:26:07:f0:6a (public)
[00:00:00.355,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0026, 0
[00:00:00.355,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x2126
```
Please try updating both Zephyr and the toolchain, also running `west update` as stm32 HAL changes may be relevant.

ABOSTM, 2020-08-05:
I also could not reproduce the issue on referenced commit 7da5f15 on Windows environment

wojciiii, 2020-08-05:
I will update zephyr/toochain and try again.

wojciiii, 2020-08-05:
Thank you for your time.

I updated both zephyr (with west update) and the sdk (zephyr-sdk-0.11.4). I verified that the new SDK is being used (by looking at verbose ninja output when building the sample).

peripheral_hids crashes as before. I'm at e73e93215381d97b3c916ed6120cdc5f7620a4cd.

I would like to find out what is going on but I need to know if there is anything else that I'm supposed to flash (I use a device with Bluetooth firmware flashed to it and this works fine as I can use Bluetooth without crashes).

wojciiii, 2020-08-06:
> I also could not reproduce the issue on referenced commit [7da5f15](https://github.com/zephyrproject-rtos/zephyr/commit/7da5f15e1577755b4d01dc05b8fac86f2da8cba8) on Windows environment

What board do you use?

I'm using a board with 256 KiB of flash. This is perhaps the source of my issues. :)

I think that the OS assumes that there are 1 MiB of flash available on this board.

                /* Set all partitions with first 812K of flash */
                /* last 212K are reseved for M0 usage */
                /* Configure partitions to make use of the whole 812K */

According to 
[Release Notes for STM32WB Copro Wireless Binaries.pdf](https://github.com/zephyrproject-rtos/zephyr/files/5032834/Release.Notes.for.STM32WB.Copro.Wireless.Binaries.pdf) there are 3 different versions of this board with different amounts of flash. 
They are flashed differently (different locations in flash the firmware images are written to).

How is this supposed to be handled in Zephyr?

When I look at https://www.st.com/en/evaluation-tools/p-nucleo-wb55.html I see the HW that I'm using.

wojciiii, 2020-08-06:
Hi,

It looks like I'm using a slightly different board which works with the Zephyr "nucleo_wb55rg" board definition.

I will close this issue since Zephyr works as expected. :)

wojciiii, 2020-08-06:
I found the part number on my board. Its STM32WB55RGV6U which has 1 MiB flash. This board should be supported by Zephyr.

wojciiii, 2020-08-06:
I flashed the board with:
stm32wb5x_FUS_fw_1_0_2.bin at 0x080EC000
stm32wb5x_BLE_Stack_full_fw.bin at 0x080CB000
using STM32CubeProgrammer.

pabigot, 2020-08-06:
That's the board I used, but I don't have those firmware images nor the STM32CubeProgrammer, so I can't diagnose this further.  It does appear to be a problem with interaction between Zephyr and that firmware.

ABOSTM, 2020-08-06:
@wojciiii ,
Yes, the MB1355C is the board supported by zephyr as nucleo_wb55rg.
And it is the same than mine.
Like you, I reflash FUS binary at 0x080EC000 (it is better to flash the binary without version as it is the latest version see https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Projects/STM32WB_Copro_Wireless_Binaries/STM32WB5x/Release_Notes.html) in my case it is v1.1.0
I also flashed stm32wb5x_BLE_Stack_full_fw.bin at 0x080CB000, version is V1.8.0 corresponding to latest GitHub stm32Cube release version (https://github.com/STMicroelectronics/STM32CubeWB/).
And I have no issue.
```
*** Booting Zephyr OS build zephyr-v2.3.0-1578-g7d920ba39b41  ***
[00:00:00.013,000] <inf> fs_nvs: 3 Sectors of 4096 bytes
[00:00:00.013,000] <inf> fs_nvs: alloc wra: 1, ad8
[00:00:00.013,000] luetooth initialized
[0m<inf> fs_Advertising successfully started
nvs: data wra: 1, 520
[00:00:00.035,000] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.038,000] <inf> bt_hci_core: Identity: 80:e1:26:00:44:3e (public)
[00:00:00.038,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0049, manufacturer 0x0030
[00:00:00.038,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x2149
```
[zephyr.zip](https://github.com/zephyrproject-rtos/zephyr/files/5036064/zephyr.zip)
You can try to flash my application binary (attached) which is samples/bluetooth/peripheral_hids with no modification, no additional debug, no additional CONFIG ... simply out of the box.
Note: make sure to have no shield and no wire on your board.
If necessary restore jumper on the board like default (see https://www.st.com/resource/en/user_manual/dm00517423-bluetooth-low-energy-and-802154-nucleo-pack-based-on-stm32wb-series-microcontrollers-stmicroelectronics.pdf). Also restore Option Bytes if you modify them.
Aim it to restore your board like it was out of the box. 

Note: I will not be able to help you further for the next 2 weeks (vacations), but you can try whether other samples are properly working (hello_word, blinky, ...)

wojciiii, 2020-08-07:
> @wojciiii ,
> Yes, the MB1355C is the board supported by zephyr as nucleo_wb55rg.
> And it is the same than mine.
> Like you, I reflash FUS binary at 0x080EC000 (it is better to flash the binary without [...]

Thanks for your help. :)

To *me* it looks like the sequence of binary firmware writes matters. 

Writting stm32wb5x_FUS_fw.bin first and then stm32wb5x_BLE_Stack_full_fw.bin resolves my issue. Writing the firmware the other way around triggers the problem this bug is about.

So this is not a Zephyr problem as such and can be closed.
I will ask about this on ST forums.

closed 2020-08-07 by wojciiii