zephyrproject-rtos/zephyr #27352: Verify that single-threaded can be made to work on all architectures

pabigot, 2020-08-03:
This is either a bug or an enhancement, partly depending on how #9808 gets settled.

Assuming that single-thread support is something we want we need to verify that it can work across architectures.

#26796 identified failure to detect interrupts on Nordic platforms.  https://github.com/zephyrproject-rtos/zephyr/issues/26796#issuecomment-668009568 confirms this is true of all Cortex-M platforms and #27343 fixes some of the issues, though not the one identified in #8393.

Using #27136 I've confirmed that rv32m1_vega_ri5cy also does not work with `CONFIG_MULTITHREADING=n`: the board hangs after the initial loop.

pabigot, 2020-08-05:
As I mentioned in triage, the purpose of this issue is to identify the impact of `CONFIG_MULTITHREADING=n` on functional capabilities of Zephyr, originally based on architecture but also functional areas like timers, I2C, SPI.  We need to know what *can* be made to work with that configuration, then assess whether the breadth of applications that could be supported warrants the complexity of testifying and maintaining code that works both with and without threads.

As an example of the limitations I noted that timers may not work.  Extending `hello_world` to this code:

```
	printk("Hello World! %s\n", CONFIG_BOARD);
	while (true) {
		k_sleep(K_SECONDS(1));
		printk("hello %u\n", k_uptime_get_32());
	}
```
on Nordic hardware by default will produce:

```
Hello World! nrf52840dk_nrf52840
hello 1007
hello 2008
hello 3009
hello 4010
hello 5012
```

displaying at nominally 1 s intervals.  When `CONFIG_MULTITHREADING=n` instead it produces:
```
Hello World! nrf52840dk_nrf52840
hello 7
hello 8
hello 9
hello 9
hello 10
hello 11
hello 12
hello 13
hello 14
```
printing as fast as possible.

This suggests that the system time infrastructure is not fully functional with multithreading disabled.

I'm skeptical that saying "You can use `CONFIG_MULTITHREADING=n`, but you won't have `k_sleep` or `k_timer` support" will be acceptable to many people.  Code that has to delay would have to be updated to divert to `k_busy_wait()` in that environment.

@carlescufi @MaureenHelm

carlescufi, 2020-08-05:
>I'm skeptical that saying "You can use CONFIG_MULTITHREADING=n, but you won't have k_sleep or k_timer support" will be acceptable to many people. Code that has to delay would have to be updated to divert to k_busy_wait() in that environment.

I think that expectations are different for every single user of `CONFIG_MULTITHREADING=n`. I have never expected (almost) anything starting with `k_` to work in that case. For me it's a basic framework to build upon when doing bare metal development. In particular MCUBoot, the biggest user of this feature, has never required nor used timeouts or sleep.
But that's why it's crucial to define the full scope of this Kconfig option, so that we are all on the same page.

nashif, 2024-02-08:
We have this alrady used on multiple architecture with tests.

closed 2024-02-08 by nashif