zephyrproject-rtos/zephyr #27362: cannot move to 1M baud rate in bt_shell

sandeep-airatechnology, 2020-08-03:
Hi,
I am using the /test/bluetooth/shell/ example.
I am trying to move to 1M baud rate and it doesn't work. However I am able to lower baud rates i.e 9600 and 115200.
I am using screen in MacOS to talk to the device.
Can you please tell me if Im missing any config? 

My overlay file looks like

> /* SPDX-License-Identifier: Apache-2.0 */
> 
> &uart0 {
> 	compatible = "nordic,nrf-uart";
> 	current-speed = <1000000>;
> 	status = "okay";
> 	hw-flow-control;
> };

carlescufi, 2020-08-04:
I just tested this with this diff:

```
diff --git a/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts b/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
index e021950552..ee69b9f320 100644
--- a/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
+++ b/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
@@ -142,7 +142,7 @@
 &uart0 {
        compatible = "nordic,nrf-uart";
        status = "okay";
-       current-speed = <115200>;
+       current-speed = <1000000>;
        tx-pin = <6>;
        rx-pin = <8>;
        rts-pin = <5>;
```

on Linux with minicom, and it worked fine. I don't think this is a bug, please try this patch instead of using an overlay. Otherwise this might be a macOS issue.

closed 2020-08-04 by carlescufi

sandeep-airatechnology, 2020-08-05:
@carlescufi Thank-you.. it works using pyserial but not using screen(I don't know why)..