zephyrproject-rtos/zephyr #27363: kernel.device.pm failed on mec15xxevb_assy6853 board.

chen-png, 2020-08-04:
**Describe the bug**
building tests/kernel/device/ on mec15xxevb_assy6853 board, **it would report a build error.**

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b mec15xxevb_assy6853 -p auto tests/kernel/device/ -**DCONFIG_DEVICE_POWER_MANAGEMENT=y**
2. see error

**Logs and console output**
zephyrproject/zephyr/subsys/power/device.c: In function 'sys_pm_create_device_list':
zephyrproject/zephyr/subsys/power/**device.c:163:22: error: 'core_devices' undeclared (first use in this function); did you mean 'pm_devices'?**
  163 |  num_pm = ARRAY_SIZE(core_devices);
      |                      ^~~~~~~~~~~~
../include/sys/util.h:67:66: note: in definition of macro 'ZERO_OR_COMPILE_ERROR'
   67 | #define ZERO_OR_COMPILE_ERROR(cond) ((int) sizeof(char[1 - 2 * !(cond)]) - 1)
      |                                                                  ^~~~
../include/sys/util.h:98:11: note: in expansion of macro 'IS_ARRAY'
   98 |  ((long) (IS_ARRAY(array) + (sizeof(array) / sizeof((array)[0]))))
      |           ^~~~~~~~
zephyrproject/zephyr/subsys/power/device.c:163:11: note: in expansion of macro 'ARRAY_SIZE'
  163 |  num_pm = ARRAY_SIZE(core_devices);
      |           ^~~~~~~~~~
zephyrproject/zephyr/subsys/power/device.c:163:22: note: each undeclared identifier is reported only once for each function it appears in
  163 |  num_pm = ARRAY_SIZE(core_devices);
      |                      ^~~~~~~~~~~~
../include/sys/util.h:67:66: note: in definition of macro 'ZERO_OR_COMPILE_ERROR'
   67 | #define ZERO_OR_COMPILE_ERROR(cond) ((int) sizeof(char[1 - 2 * !(cond)]) - 1)
      |                                                                  ^~~~
../include/sys/util.h:98:11: note: in expansion of macro 'IS_ARRAY'
   98 |  ((long) (IS_ARRAY(array) + (sizeof(array) / sizeof((array)[0]))))
      |           ^~~~~~~~
zephyrproject/zephyr/subsys/power/device.c:163:11: note: in expansion of macro 'ARRAY_SIZE'
  163 |  num_pm = ARRAY_SIZE(core_devices);
      |           ^~~~~~~~~~
At top level:
zephyrproject/zephyr/subsys/power/device.c:68:21: warning: 'pm_devices' defined but not used [-Wunused-variable]
   68 | static device_idx_t pm_devices[MAX_PM_DEVICES];
      |                     ^~~~~~~~~~
[60/111] Building C object zephyr/drivers/adc/CMakeFiles/drivers__adc.dir/adc_mchp_xec.c.obj
ninja: build stopped: subcommand failed.

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.11.4
 - Commit ID: 4137d3380f

dcpleung, 2020-08-04:
You missed this part of the error log:
```
zephyr/subsys/power/device.c:53:2: error: #error "Add SoC's core devices list for PM"
   53 | #error "Add SoC's core devices list for PM"
      |  ^~~~~
```

We have never enabled device power management on MEC1501 so this is expected behavior. Closing...

closed 2020-08-04 by dcpleung

chen-png, 2020-08-13:
> We have never enabled device power management on MEC1501 so this is expected behavior. Closing...

for this test case, should we exclude mec board?

dcpleung, 2020-08-13:
The issue is due to your explicit enabling of `-DCONFIG_DEVICE_POWER_MANAGEMENT=y` on a board that *does not* have device power management enabled. There are no CI errors on this, so there is no need to exclude this board in the yaml file.

Zhaoningx, 2020-08-14:
but this tese case will be  selected by sanitycheck script to run on mec15xxevb_assy5853 board, so it should be excluded in yaml file.