zephyrproject-rtos/zephyr #27366: tests: net: regression on many tests

ABOSTM, 2020-08-04:
**Describe the bug**
Following tests, which used to be passed, are now failed on automatic bench (nucleo_f746zg, nucleo_f429zi)
* tests/net/utils/net.util
* tests/net/socket/getnameinfo/net.socket.get_name_info
* tests/net/socket/select/net.socket.select
* tests/net/socket/misc/net.socket.misc
* tests/net/socket/udp/net.socket.udp
* tests/net/socket/socketpair/net.socket.socketpair
* tests/net/socket/getaddrinfo/net.socket.get_addr_info
* tests/net/socket/tcp/net.socket.tcp
* tests/net/socket/net_mgmt/net.socket.mgmt
* tests/net/socket/af_packet/net.socket.packet
* tests/net/lib/mqtt_packet/net.mqtt.packet
* tests/net/ptp/clock/net.ptp.clock
* tests/net/iface/net.iface

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_f429zi tests/net/utils/ 
2. west flash
3. See error

**Expected behavior**
Tests passed

**Logs and console output**
```
E: HAL_ETH_Init Timed out
ASSERTION FAIL [!sys_dnode_is_linked(&to->node)] @ WEST_TOPDIR/zephyr/kernel/timeout.c:103

E: r0/a1:  0x00000004  r1/a2:  0x00000067  r2/a3:  0x40004800
E: r3/a4:  0x00000002 r12/ip:  0x00000000 r14/lr:  0x08010635
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x08015bfa
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x2000f000 (unknown)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux and  Windows
 - Toolchain : Zephyr SDK
 - Commit SHA : e90873f290e2351232358239152a0fbfde83ea7e

**1st Analysis**
After a bisect, I found that regression occurs after merge of PR #23149 
So it is passed with ec5823c1ea810f69473346d2a8f845b37ab34515 and failed with e90873f290e2351232358239152a0fbfde83ea7e
Note : The 1st message in the console` E: HAL_ETH_Init Timed out` is normal (just because there is no ethernet cable connected) and it appears when test is successful too.
I could not make direct link between PR #23149 and network tests tests/net/xxx.
Also looking at #PR#23149, I see no link with STM32 boards.

erwango, 2020-08-10:
Guilty commit points to https://github.com/zephyrproject-rtos/zephyr/pull/24714 which does not impact directly stm32 target but has wide kernel impacts.
Need to check if only stm32 targets are impacted.

erwango, 2020-08-10:
Tried some net tests on latest master (zephyr-v2.3.0-1729-g4a72108be8) and they are passed (while I manage to reproduce the issue on reported guilty commit.
Doing wider testing to confirm the issue has been solved meanwhile.

erwango, 2020-08-10:
Issue solved on latest master (sanitycheck --device-testing -T tests/net -p nucleo_f429zi -p nucleo_f746zg  => all passed)

closed 2020-08-10 by erwango