zephyrproject-rtos/zephyr #27367: Sprintf -  error while sending data to SD card

EdytaBosacka, 2020-08-04:
While I use sprintf function and I want to send some data to .csv file to SD card I get this error: "******USAGE FAULT******"
                "Illegal load of EXC_RETURN into PC"

When I  send data to SD card and I dont use "sprintf" function, the problem doesn't occur.  Also when I only use sprintf function without using SD card, this function works.
But somehow combining sprintf function with SD card  causes an error :( 

`#include <zephyr.h>
#include <device.h>
#include <devicetree.h>
#include <drivers/gpio.h>
#include <string.h>
#include <stdio.h>
#include <kernel.h>
#include <stdio.h>
#include <disk/disk_access.h>
#include <logging/log.h>
#include <fs/fs.h>
#include <ff.h>

#define BUF_MAXSIZE	256

typedef unsigned char       u8_t;
typedef unsigned int        u32_t;
static uint8_t rx_buf[BUF_MAXSIZE];

static FATFS fat_fs;

LOG_MODULE_REGISTER(main);
/* mounting info */
static struct fs_mount_t mp = {
	.type = FS_FATFS,
	.fs_data = &fat_fs,
};

int fs_register(enum fs_type type, struct fs_file_system_t *fs);

static const char *disk_mount_pt = "/SD:";

struct device *uart1_dev;

int pozycja = 0;

void main(void)
{
	
    mp.mnt_point = disk_mount_pt;
    int res = fs_mount(&mp);
    char linia[20];

while(1){

    struct fs_file_t zfp;
    int ok = fs_open(&zfp,"/SD:/plink.csv");
    
  
    // if(ok!=0){
    //     printk("Error opening file %s [%d]\n", "/SD:/plink.csv", res);
    //     return res;
    //    };
   
    
    
    // int os_y=2;
    // int os_z=3;
    fs_seek(&zfp, pozycja, FS_SEEK_SET);
  
    int os_x=1;
    int wynik = sprintf(linia,"%x \n", os_x);
    printk("%s",linia);

    //printk("%p wynik=%d",linia,wynik);
    int zapis = fs_write(&zfp, linia, 5); 
    pozycja = pozycja + zapis;
    
    fs_close(&zfp);
   

}
  

}
`

MaureenHelm, 2020-08-04:
Could you please provide some more information about what board and zephyr version you are using?

EdytaBosacka, 2020-08-05:
@MaureenHelm  My board: nRF52840-dk
                             Zephyr version: 2.3.0

nvlsianpu, 2020-09-14:
@EdytaBosacka At the first glance this looks like a stack to small issue. Pleas Increase thread stack size.

closed 2020-11-28 by github-actions[bot]