zephyrproject-rtos/zephyr #27369: spi: stm32: dma: rx transfer error when spi_write called

macsiahid, 2020-08-04:
**Describe the bug**
It is not possible to receive data after call to spi_write.
It happens for SPI with DMA on STM32WB55 MCU.
When only spi_transceive is used then communication works fine. 

**Description of flow causing problems**
After calling spi_write, transfer error flag (TEIF) in DMA_ISR register is set for channel corresponding to SPI_RX and next this channel is disabled.  It may be caused by assigning NULL to DMA_CMARx as an address of Rx buffer (which is the case for spi_write). Moreover, error flag (TEIF) cannot be cleared because transfer error interrupt (TEIE bit in DMA_CCRx) is not enabled and even inside body of dma_stm32_irq_handler there is no call for clearing this flag. 
Another thing is that inside the dma_stm32_configure function, the "true" value is assigned to stream->busy flag (struct dma_stm32_stream *stream) and this flag cannot be cleared (set to "false") anymore because dma_stm32_irq_handler will be not called. 
Because that busy flag is set that’s why configuring DMA channel (corresponding to Rx) again is not possible (see dma_stm32_configure(), dma_stm32.c l. 260).  

**To Reproduce**
Configure SPI to work with DMA
Use spi_write() to send some data to external module.
Next use spi_write() (or spi_transceive()) to notice that configuring Rx again is not possible.

**Impact**
Cannot communicate with external modules via SPI with use of DMA.

**Screenshots or console output**

**Environment**
OS: Linux
Toolchain: Zephyr SDK 0.11.3
Commit: 6c7a387

**Additional context**
Bug was found during communicating via SPI with external flash using spi_flash_w25qxxdv driver provided by Zephyr.
Driver uses both spi_write() and spi_transceive() during operation.

ABOSTM, 2020-08-05:
@macsiahid, 
SPI with DMA has been successfully tested on nucleo_wb55rg with :
`tests\drivers\spi\spi_loopback`
So can you test the same ?

This tests defines some devicetree overlay : 
`tests\drivers\spi\spi_loopback\boards\nucleo_wb55rg.overlay`
Please verify you also define updated you devicetree in your project.

the same for CONFIG switches:
` tests\drivers\spi\spi_loopback\boards\nucleo_wb55rg.conf`

macsiahid, 2020-08-05:
@ABOSTM,
I've already run tests located at tests\drivers\spi\spi_loopback on nucleo_wb55rg and they've all passed.
I saw there was no test using spi_write(), only spi_transceive().
In order to reproduce the bug I noticed during using  spi_flash_w25qxxdv driver, I've run spi_loopback tests
with code added:

    static int spi_write_test(struct device *dev, struct spi_config *spi_conf)
    {
        uint8_t buffer_tx = 0x05;
        int ret;
      
        struct spi_buf buf = {
            .buf = &buffer_tx,
            .len = 1
        };
      
        struct spi_buf_set tx = {
            .buffers = &buf,
            .count = 1
        };
     
        if (IS_ENABLED(CONFIG_SPI_STM32_DMA)) {
            LOG_INF("DMA enabled");
        }
      
        /* 
         * Note: Please remember that implementation for spi_write() (zephyr/include/drivers/spi.h) is: 
         *  
         * static inline int spi_write(struct device *dev
         *                             const struct spi_config *config,
         *                             const struct spi_buf_set *tx_bufs)
         *
         *     return spi_transceive(dev, config, tx_bufs, NULL);
         * }
         */
      
        ret = spi_write(dev, spi_conf, &tx);
        LOG_INF("Code %d", ret);
      
        ret = spi_write(dev, spi_conf, &tx);
        LOG_INF("Code %d", ret);
      
        if (ret) {
            LOG_ERR("Code %d", ret);
            zassert_false(ret, "SPI write failed");
            return -1;
        }
      
        return 0;
    }

This test fails because of reasons described at **Description of flow causing problems**

ABOSTM, 2020-08-05:
@macsiahid 
I am not clear about your use case. As I understand you can reproduce the issue by :
starting from  tests\drivers\spi\spi_loopback
and add the function above `spi_write_test()`
But I guess there are some more modifications , at least to call this new function...
Are the 2 calls to `spi_write()` working ?
Also you said you cannot read data after` spi_write().` Does it means that this is not  `spi_write_test()`  which fails, but the later functions of original test ? (meaning that place where you called  `spi_write()` is important)
So can you please send the whole file, so we are working on the same base ?

macsiahid, 2020-08-06:
@ABOSTM,
I've added spi_write_test() and made a little modification in the code to call it as you can see in the attached file.
[spi.zip](https://github.com/zephyrproject-rtos/zephyr/files/5033117/spi.zip)

As I described earlier after first call of spi_write() transfer error flag (TEIF) in DMA_ISR register is set for channel corresponding to SPI_RX. After second spi_write() call error code is returned because DMA channel corresponding to SPI_RX is busy and DMAMUX cannot be configured. Please take a look what happens with stream->busy flag (struct dma_stm32_stream *stream) 
(see dma_stm32_configure(), dma_stm32.c l. 260) during the whole process.

ABOSTM, 2020-08-06:
@macsiahid ,
Ok I reproduce your issue, and yes it seems there is RX DMA which is started and which should not.
But I think it is not the only problem:
Even if I prevent driver from starting RX DMA, I still have an issue because the next transceive (Read + write) which arrives after spi_write(), get as 1st RX data values, the data that where present during both spi_write transfert (ie 0x05 0x05 due to loopback). It is probably due to the hardware fifo that keep those values.

So, in order to quickly solve your issue, I propose a workaround:
use in your application `spi_transceive()` instead of` spi_write()`, and simply discard the RX data you don't want.

@FRASTM
1) in file spi_ll_stm32.C,  function spi_dma_move_buffers() , is it normal to call function `spi_stm32_dma_rx_load() ` in the else ?
 ```
	if (data->ctx.rx_len) {
		ret = spi_stm32_dma_rx_load(dev, data->ctx.rx_buf,
					    data->dma_segment_len);
	} else {
		ret = spi_stm32_dma_rx_load(dev, NULL, data->dma_segment_len);
	}
``` 
because it starts RX DMA when RX length is null.
Note: I comment the 'else' part and this avoid issue described in this issue. There is no more DMA stuck.
(but I face new issue in further data received see 2) point)

By symmetry, why do we call spi_stm32_dma_tx_load()  when data->ctx.tx_len is 0 ?
 
 
2) In order get ride of unwanted RX data (which are probably in the hardware fifo), I tried to flush the fifo, using in transceive_dma() the same code than in transceive() :
```
#if DT_HAS_COMPAT_STATUS_OKAY(st_stm32_spi_fifo)
	/* Flush RX buffer */
	while (ll_func_rx_is_not_empty(spi)) {
		(void) LL_SPI_ReceiveData8(spi);
	}
#endif
```
but it doesn't work: ll_func_rx_is_not_empty(spi) is always false.
 
 
3) why do we skip test "every 4" when DMA is enabled ? 
```
	if (IS_ENABLED(CONFIG_SPI_STM32_DMA)) {
		LOG_INF("Skip every 4");
		return 0;
```
Does it means that result of API spi_transceive() depends on DMA activation ? that sound abnormal to me.

4) Similarly to this use case, I wonder what would happend if there are several call to spi_transceive() but with RX buffer length smaller than TX buffer length ? (with or without DMA)

lowlander, 2020-08-08:
I also had problems with the STM32 SPI DMA (on a stm32f4). I created a PR #27263 which fixes things for me, maybe it also helps in this case.

erwango, 2020-09-03:
> I also had problems with the STM32 SPI DMA (on a stm32f4). I created a PR #27263 which fixes things for me, maybe it also helps in this case.

@macsiahid can you check this possibly fixes the issue you're reporting ?

lowlander, 2020-09-04:
PR #27263 landed in master, @macsiahid could you please try the latest master?

Zzzwierzak, 2020-10-01:
@erwango  , @lowlander   I've tested it on behalf of @macsiahid, it works on latest.

erwango, 2020-10-01:
@Zzzwierzak Thanks for this feedback! Can we close the point then ?

Zzzwierzak, 2020-10-01:
@erwango Yes, you can close it.

closed 2020-10-01 by erwango