zephyrproject-rtos/zephyr #27380: Cannot use mcuboot with i.MXRT1060 due to a problem with the vector table address

armandciejak, 2020-08-05:
**Describe the bug**
When using Zephyr with mcuboot on the MIMXRT1060-EVK the system does not boot correctly. mcuboot executes fine and starts Zephyr but the Zephyr's initialization does not complete and the system ends up in the NMI handler.

**To Reproduce**
See https://community.nxp.com/message/1342301

**Expected behavior**
I expect to be able to use mcuboot with the i.MXRT1060.

**Impact**
Not able to use mcuboot.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: SDK 0.11.3 with Zephyr 2.3

**Additional context**
See https://community.nxp.com/message/1342301

** Workaround **

I changed the value of `TEXT_SECTION_OFFSET` from 0x200 to 0x100.

armandciejak, 2020-08-05:
@MaureenHelm Could you please escalate this problem at NXP?

MaureenHelm, 2020-08-05:
How have you configured the flash partitions for mcuboot? There aren't any partitions defined upstream yet for this board.

> I changed the value of `TEXT_SECTION_OFFSET` from 0x200 to 0x100.

Are you missing a digit in these values? The default value is 0x2000 not 0x200.

https://github.com/zephyrproject-rtos/zephyr/blob/b8c78e254ff875680e99c9f131fbe285c4575927/soc/arm/nxp_imx/rt/Kconfig.defconfig.series#L11-L12

armandciejak, 2020-08-06:
Hi @MaureenHelm I didn't describe the situation completely and I mixed stuff from eval board and our custom board based on the i.MXRT1060.

On our custom board we use mcuboot, therefore the `TEXT_SECTION_OFFSET` is set to 0x200 for the Zephyr application which causes the problem described here https://community.nxp.com/message/1342301, to work around the problem we changed the value to 0x100.

In order to reproduce the problem on the eval board without mcuboot, for NXP to find the root cause of the problem, I changed the `TEXT_SECTION_OFFSET` to 0x2200, with this value the problem occurs as well.

I think there is a bug on the CPU, that's why I originally asked the question on the NXP forum.

MaureenHelm, 2020-08-13:
Thanks @armandciejak for the additional information, that helps a lot and I'm now able to reproduce the issue. I haven't gotten to the bottom of it yet, but I wanted to let you know that I am looking at it. The zephyr release team categorized this as a low priority issue because it affects one platform, but that doesn't mean it isn't important. Will get back to you again soon.

MaureenHelm, 2020-08-13:
The Cortex-M7 core requires that the vector table be aligned as follows:
```
When setting TBLOFF, you must align the offset to the number of exception entries in the vector
table. The minimum alignment is 32 words, enough for up to 16 interrupts. For more interrupts,
adjust the alignment by rounding up to the next power of two. For example, if you require 21
interrupts, the alignment must be on a 64-word boundary because the required table size is 37
words, and the next power of two is 64.
```
https://developer.arm.com/documentation/dui0646/c/cortex-m7-peripherals/system-control-block/vector-table-offset-register?lang=en

The RT1060 vector table has 16+160 entries, therefore it must be aligned to a 0x400 byte boundary. Setting `CONFIG_TEXT_SECTION_OFFSET=0x2200` doesn't satisfy this alignment.

There was a recent change in Zephyr after 2.3 that helped diagnose this issue, see #26800

armandciejak, 2020-08-17:
@MaureenHelm I do not agree with your conclusion on the vector table offset alignment. The RT1060 vector table has 176 entries and the next power of 2 is 0x100 and not 0x400, which means the 0x200 offset is correct. 

On my board and the MIMXRT1060-EVK I tried with both `CONFIG_TEXT_SECTION_OFFSET=0x2100` and `CONFIG_TEXT_SECTION_OFFSET=0x2400` and they both works fine while `CONFIG_TEXT_SECTION_OFFSET=0x2200` fails.

That's why I suspect a bug in the silicon.

utzig, 2020-08-17:
> I do not agree with your conclusion on the vector table offset alignment. The RT1060 vector table has 176 entries and the next power of 2 is 0x100 and not 0x400, which means the 0x200 offset is correct.

Each vector table entry is 4-byte long so 0x400 is the correct value.

armandciejak, 2020-08-17:
> Each vector table entry is 4-byte long so 0x400 is the correct value.

I missed that part. Thanks for the clarification.

closed 2020-08-17 by MaureenHelm