zephyrproject-rtos/zephyr #27403: uart_fifo_read can only read one character

1999kevin, 2020-08-06:
**Describe the bug**

int uart_fifo_init(void){
	uint8_t c;
	dev = device_get_binding("UART_2");
	if (dev == NULL) {
		printk("cannot find device\n");
		return -1;
	}
	uart_irq_rx_disable(dev);
	uart_irq_tx_disable(dev);	
	uart_irq_callback_set(dev, uart_fifo_callback);
	/* Drain the fifo */
	if (uart_irq_rx_ready(dev)) {
		while (uart_fifo_read(dev, &c, 1)) {
			;
		}
	}
	uart_irq_rx_enable(dev);
	/* Enable all interrupts unconditionally. Note that this is due
	 * to Zephyr issue #8393. This should be removed once the
	 * issue is fixed in upstream Zephyr.
	 */
	irq_unlock(0);
	return 0;
}
void uart_fifo_callback(struct device *dev){
	while(uart_irq_update(dev) && uart_irq_is_pending(dev)){
		if (!uart_irq_rx_ready(dev)){
			continue;
		}
		else{
			buf.size += uart_fifo_read(dev, &buf.buf[buf.size], 10);
			printk("buf.size:%d\n",buf.size);
		}
		printk("while1\n");
	}
}

when I set up the callback function as above and input a line of characters by serial to uart2 rx, it can only read the first character 

**Expected behavior**
A clear and concise description of what you expected to happen.
If I input "aaa" to uart2, my buf should contain 0x61,0x61,0x61. But now it only contains one 0x61.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)
[image]

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: gcc-arm-none-eabi

1999kevin, 2020-08-06:
I run this program in stm32_min_dev_blue

ABOSTM, 2020-08-06:
@carlescufi please stop assigned me issue, I am about to leave for 2 weeks vacations :-)
I hand over the baton to @erwango , coming back next week

1999kevin, 2020-08-07:
@erwango could you help me figure out why this bug happen? Thank you very much!

1999kevin, 2020-08-14:
solved it by removing printk() command in the callback function

closed 2020-08-14 by 1999kevin