zephyrproject-rtos/zephyr #27404: Is DT_INST_FOREACH_STATUS_OKAY broken on v2.3?

armingrundmann, 2020-08-06:
I am experimenting with implementing a custom device driver using devicetree. There I encountered the macro `DT_INST_FOREACH_STATUS_OKAY(fn)`. And despite my bindings and devicetree + overlays being correct, nothing happens.

I then managed to track down the issue: `DT_INST_FOREACH_STATUS_OKAY` uses `COND_CODE_1`, which conditionally expands, depending on whether the first parameter is `1` or not. As first parameter `DT_HAS_COMPAT_STATUS_OKAY` is used, which is supposed to expand to `1` if there is at least one compatible device on the devicetree. `DT_HAS_COMPAT_STATUS_OKAY` itself uses `IS_ENABLED` and thus actually expands to `true` or `false`. Thus `COND_CODE_1(DT_HAS_COMPAT_STATUS_OKAY(...), ("yay"), ("nay"))` always expands to "nay".

If I `#define _XXXXtrue _YYYY,` beforehand, it works as expected.

Minimal working example:
```
#include <zephyr.h>

//#define _XXXXtrue _YYYY,    // uncomment this for "yay"

// using c++11 constexpr here for convenience
constexpr const char* test = COND_CODE_1(\
	DT_HAS_COMPAT_STATUS_OKAY(nordic_nrf_spi),\ // choose anything available on your devicetree
	("yay"),\
	("nay")
);
// test gets initialized as "nay"

void main() { }
```

I am on commit c0d3c8004422b359fd2b6257a2c8e226eff54dd6(current head of v2.3-branch). I am developing using a cmake based vscode setup.

Am I doing something wrong, or must this be fixed?

Best regards,
Armin

mbolivar-nordic, 2020-08-06:
```
// using c++11 constexpr here for convenience
```

Zephyr's minimal libc defines true to 1 and false to 0 unless C++ is being used. So my guess is that this just doesn't work in C++ and it's actually a sys/util.h bug, not a devicetree.h bug.

mbolivar-nordic, 2020-08-06:
@nordic-krch I think IS_ENABLED() internals may need a fix for C++ as described above. What do you think?

pabigot, 2020-08-11:
I don't think `IS_ENABLED()` is the problem, rather in sys/util.h:
```
#define IS_EMPTY(a) Z_IS_EMPTY_(a, true, false,)
```
which is not going to work.  I'll reproduce and resolve this if it's possible.

PR #27502 (2020-08-12): sys/util: fix macro support when building with C++
true and false in a C environment are macros that are defined to 1 and 0 respectively.  In C++ they are keywords, and as such cause failures when macro tests require they expand to a 1 or 0 token.  Use the actual token values rather than the macro names when defining the macros.

Also fix a C++ issue in ztest, and rework the util test to do tests in both C and C++ environments.

Fixes #27404.
closed 2020-08-12 by MaureenHelm