zephyrproject-rtos/zephyr #27420: drivers.uart: config test failure on twr_kv58f220m platform

hakehuang, 2020-08-07:
**Describe the bug**
  Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_basic_api/src/test_uart_config.c:84: test_config_get: (ret == 0 is false)

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=twr_kv58f220m ..
3. make flash

**Expected behavior**
all cases pass

**Impact**
uart config

**Logs and console output**

```
*** Booting Zephyr OS version 2.3.99  ***
Running test suite uart_basic_test
===================================================================
START - test_uart_configure
 PASS - test_uart_configure
===================================================================
START - test_uart_config_get
This is a configure_get test.

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_basic_api/src/test_uart_config.c:84: test_config_get: (ret == 0 is false)
get config error
 FAIL - test_uart_config_get
===================================================================
START - test_uart_fifo_fill
This is a FIFO test.
 PASS - test_uart_fifo_fill
===================================================================
START - test_uart_fifo_read
Please send characters to serial console
abcDEF 012345 ~!@#$%
 PASS - test_uart_fifo_read
===================================================================
START - test_uart_poll_in
Please send characters to serial console

 PASS - test_uart_poll_in
===================================================================
START - test_uart_poll_out
This is a POLL test.
 PASS - test_uart_poll_out
===================================================================
Test suite uart_basic_test failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 -  Zephyr OS version 2.3.99

hakehuang, 2020-08-07:
@MaureenHelm

galak, 2020-08-11:
Able to reproduce on FRDM-K64F as well.

PR #27498 (2020-08-13): drivers: uart_mcux: Implement uart_config_get API
When uart_configure support was added uart_config_get was not
implemented.  Move the 'struct uart_config' to uart_mcux_data lets us
save the uart_config settings and easily return it in
uart_mcux_config_get.

Fixes: #27420
commit 989ee987 (2020-08-13): drivers: uart_mcux: Implement uart_config_get API

When uart_configure support was added uart_config_get was not
implemented.  Move the 'struct uart_config' to uart_mcux_data lets us
save the uart_config settings and easily return it in
uart_mcux_config_get.

Fixes: #27420
closed 2020-08-13 by carlescufi