zephyrproject-rtos/zephyr #27446: Unable to flash cc1352r (no xds) with openocd in Zephyr SDK

cfriedt, 2020-08-08:
**Describe the bug**

I'm currently targeting the `BOARD=cc1352r_sensortag` (for which I'll make a PR shortly), but the same issue should be reproducible with `BOARD=cc1352r1_launchxl`.

The only JTAG device I have atm is a JLink Segger. I have not been able to program this device with jlink itself yet, which is why I'm using openocd.

**To Reproduce**

If using the cc1352r1_launchxl, remove the JTAG jumpers and attach directly to the 10-pin JTAG header beside BTN-2.

1. Put the following into `${BOARD}/support/openocd.cfg`
    ```
    source [find interface/jlink.cfg]
    adapter_khz 4000
    transport select jtag
    reset_config trst_only
    source [find target/ti_cc13x2.cfg]
    ```
1. Put the following inside ~/.zephyrrc
    ```
    export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
    export ZEPHYR_SDK_INSTALL_DIR=~/zephyr-sdk-0.11.3
    export BOARD=cc1352r_sensortag
    export ZEPHYR_PROJECT=samples/hello_world
    ```
1. source zephyr-env.sh
1. west build -p always $ZEPHYR_PROJECT -t flash
1. See error

**Expected behavior**
Flash the chip. Wait for debug.

**Impact**
Makes development annoying / frustrating.

**Logs and console output**
```
$ west build $ZEPHYR_PROJECT -t flash
-- west build: running target flash
[0/1] Flashing cc1352r_sensortag
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/cfriedt/workspace/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-01341-g580d06d9d-dirty (2020-05-16-15:41)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter_nsrst_delay: 100

Info : J-Link V10 compiled Aug 27 2019 09:24:55
Info : Hardware version: 10.10
Info : VTarget = 2.651 V
Info : clock speed 4000 kHz
Info : JTAG tap: cc13x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc13x2.cpu enabled
Info : cc13x2.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : cc13x2.cpu: external reset detected
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* cc13x2.cpu         cortex_m   little cc13x2.cpu         running

Info : JTAG tap: cc13x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc13x2.cpu enabled
Error: Debug regions are unpowered, an unexpected reset might have happened
Error: JTAG-DP STICKY ERROR
Error: Could not find MEM-AP to control the core
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: DP initialisation failed

FATAL ERROR: command exited with status 1: /home/cfriedt/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/cfriedt/workspace/zephyrproject/zephyr/boards/arm/cc1352r_sensortag/support -s /home/cfriedt/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/cfriedt/workspace/zephyrproject/zephyr/boards/arm/cc1352r_sensortag/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /home/cfriedt/workspace/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image /home/cfriedt/workspace/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
FAILED: zephyr/cmake/flash/CMakeFiles/flash 
cd /home/cfriedt/workspace/zephyrproject/zephyr/build && /usr/local/bin/cmake -E env /home/cfriedt/.local/bin/west flash --skip-rebuild
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/cfriedt/workspace/zephyrproject/zephyr/build --target flash
```

**Environment (please complete the following information):**
 - OS: Linux / Ubuntu (Bionic)
 - Toolchain: Zephyr SDK 0.3, Zephyr SDK 0.2
 - Commit: 970a9a02ad8a340817a4bc707b841f5e697dbd15

**Additional context**
The exact same file succeeds with openocd git
```
$ openocd -f boards/arm/cc1352r_sensortag/support/openocd.cfg -c "init" -c "reset halt" -c "flash write_image erase $PWD/build/zephyr/zephyr.elf"
Open On-Chip Debugger 0.10.0+dev-01372-gfa9a4d4db (2020-08-08-09:50)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
cortex_m reset_config vectreset

Info : J-Link V10 compiled Aug 27 2019 09:24:55
Info : Hardware version: 10.10
Info : VTarget = 2.651 V
Info : clock speed 4000 kHz
Info : JTAG tap: cc13x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc13x2.cpu enabled
Info : cc13x2.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for cc13x2.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: cc13x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc13x2.cpu enabled
Warn : Only resetting the Cortex-M core, use a reset-init event handler to reset any peripherals or configure hardware srst support.
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000be4 msp: 0x20000600
Info : Flash write discontinued at 0x00002614, next section at 0x00057fa8
Warn : Adding extra erase range, 0x00056000 .. 0x00057fa7
auto erase enabled
wrote 16472 bytes from file /home/cfriedt/workspace/zephyrproject/zephyr/build/zephyr/zephyr.elf in 1.354355s (11.877 KiB/s)

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```

Also notable, is that Chapter 6 of [the TRM](https://www.ti.com/lit/an/swra571a/swra571a.pdf) specifically says

> There is no dedicated I/O pin for TRST. The debug subsystem is reset with system-wide resets and power-on reset.

The [image] shows that there is only one pin. I'm only using what works with openocd + jlink based on trial and error (it's exhausting sometimes!!).

statropy, 2020-08-08:
I was able to program a CC1352R via J-Link by using this as the board.cmake:

`board_runner_args(jlink "--device=CC1352R1F3" "--iface=jtag" "--tool-opt=-jtagconf -1,-1 -autoconnect 1")
include(${ZEPHYR_BASE}/boards/common/jlink.board.cmake)`

It took quite a bit of trial find those settings using J-Link Commander

vanti, 2020-08-10:
I know @galak has a plan to pull in some recent improvements from TI submitted to OpenOCD git. Maybe he can comment on the status of that effort. Until then I'd expect the version on OpenOCD git to be more stable.

galak, 2020-08-10:
> I know @galak has a plan to pull in some recent improvements from TI submitted to OpenOCD git. Maybe he can comment on the status of that effort. Until then I'd expect the version on OpenOCD git to be more stable.

@cfriedt:

Here a pre-release build that has an updated openocd.  Can you test that out and let me know if there are issues:

builds.zephyrproject.org/zephyrproject-rtos/sdk-ng/258/zephyr-sdk-x86_64-hosttools-standalone-0.9.sh

cfriedt, 2020-08-11:
@MaureenHelm & @galak - thanks, I''ll give the prerelease a shot.
@statropy - thanks as well. I'll add that to #27447

cfriedt, 2020-08-11:
All of those solutions worked!

Thanks everyone!

@galak - would you prefer that I close the issue or keep it open until the prerelease is released?

galak, 2020-08-11:
> @galak - would you prefer that I close the issue or keep it open until the prerelease is released?

Lets go ahead and close the issue.

closed 2020-08-11 by cfriedt