zephyrproject-rtos/zephyr #27448: fatal error: device_imx.h: No such file or directory

jasonedn, 2020-08-08:
**Describe the bug**
When building "blinky" on MIMXRT1020_evk when PWM driver enabled. 

**To Reproduce**
Steps to reproduce the behavior:
1. Enable PWM driver on IMX
2. west build -p auto -b mimxrt1020_evk samples/basic/blinky
3. See error

**Expected behavior**
It should see "device_imx.h" that is located at zephyrproject\modules\hal\imx\devices.

**Impact**
Fail to build the example

**Logs and console output**
zephyrproject/zephyr/drivers/pwm/pwm_imx.c:10:10: fatal error: device_imx.h: No such file or directory
   10 | #include <device_imx.h>
      |          ^~~~~~~~~~~~~~

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain:  Zephyr SDK)
 - Commit: ce5951

**Additional context**
I don't expect the build to pass but want to know how to include the header files in the modules

MaureenHelm, 2020-08-10:
How did you enable the PWM driver? It looks like you have enabled PWM driver for i.MX6/7, not the driver for i.MX RT.

jasonedn, 2020-08-10:
Hello Maureen,

I included this folder in the project CMakelists.txt and this error was
fixed. I think we'd better make some changes in the CMakeLists.txt in the
hal\nxp\imx\devices, but I haven't figured it out.

As for i.MX6/7 or i.MX RT, I did select i.MX RT as shown below.

[image: image.png]

However it failed to build because i.MX RT was not supported in the
device_imx.h as below:

#ifndef __DEVICE_IMX_H__
#define __DEVICE_IMX_H__

/*
 * Include the cpu specific register header files.
 *
 * The CPU macro should be declared in the project or makefile.
 */
#if defined(CONFIG_SOC_MCIMX6X_M4)

    /* CMSIS-style register definitions */
    #include "MCIMX6X_M4.h"
    #define RDC_SEMAPHORE_MASTER_SELF   (5)
    #define SEMA4_PROCESSOR_SELF        (1)

#elif defined(CONFIG_SOC_MCIMX7_M4)

    /* CMSIS-style register definitions */
    #include "MCIMX7D_M4.h"

    #define RDC_SEMAPHORE_MASTER_SELF   (6)
    #define SEMA4_PROCESSOR_SELF        (1)

#else
    #error "No valid CPU defined!"
#endif

#endif /* __DEVICE_IMX_H__ */

MaureenHelm, 2020-08-10:
Everything in `hal/nxp/imx` is for older i.MX 6/7 parts that don't have an MCUXpresso SDK:
https://github.com/zephyrproject-rtos/hal_nxp/blob/727b36cfe9e2f2aa5eb88d608e6334c8d548cee9/imx/README#L19-L31

For i.MX RT parts you should be using `hal/nxp/mcux` instead, which is already included for RT1020. You shouldn't need to add it again.
https://github.com/zephyrproject-rtos/hal_nxp/blob/727b36cfe9e2f2aa5eb88d608e6334c8d548cee9/mcux/CMakeLists.txt#L28-L49

I think what you probably want to do is extend the FlexPWM driver from RT1064 to RT1020 by adding `HAS_MCUX_PWM` and enabling the flexpwm dts nodes.

jasonedn, 2020-08-11:
Hello Maureen,
 As you said, it was fixed by adding HAS_MCUX_PWM to RT1020.
Thanks
Jason

closed 2020-08-11 by jasonedn