zephyrproject-rtos/zephyr #27457: Add support for Nordic nrfx PDM driver for Nordic Thingy52

frederik-popp, 2020-08-10:
**Is your feature request related to a problem? Please describe.**
It would be desirable to incorporate the nRFX PDM Driver into Zephyr, similar to what has been done with other nRFX drivers.
This would allow for Zephyr to utilize the PDM microphone on the Nordic Thingy52 (nRF52_pca20020).
Initially, this support would only include the nRF52_pca20020 board, but could then later easily incorporate other compatible Nordic nRF5 boards.

**Describe the solution you'd like**
Create a generic PDM API in zephyr/drivers/include, with the functions being start and stop sampling, and a function to connect a user data handler function.
Create an implementation of this PDM API utilizing the nrfx_pdm driver found the hal module.
Updating the nRF52832.dtsi to allow support for this hardware capability.
Updating the nRF52_pca20020.dts to allow and enable support for the PDM driver.
Creating a small sample application using the implemented driver on the Nordic Thingy52.

I have already made a implemented a solution like this on an older, local version of Zephyr at work, where I was able to get a PDM sample from the microphone on the Thingy52.
I would like to do the same with the newest version of Zephyr (2.3.0), while also contributing in accordance with the Contribution Guidelines (i.e. opening a pull request from my own forked repository of Zephyr 2.3.0, once I have everything working)

MaureenHelm, 2020-08-10:
cc: @carlescufi

carlescufi, 2020-08-14:
@frederik-popp thanks for the proposal. Currently there are codec and DMIC APIs here:
https://github.com/zephyrproject-rtos/zephyr/tree/master/include/audio

Depending on what you want to do you could possibly add the proposed new API header in that folder, or use (or extend) the existing DMIC one?

>I would like to do the same with the newest version of Zephyr (2.3.0), 

When writing new functionality you really want to base yourself off of the current master, not a released version. This is described here:

https://docs.zephyrproject.org/latest/contribute/index.html#contribution-workflow

frederik-popp, 2020-08-18:
Hi @carlescufi 
> Depending on what you want to do you could possibly add the proposed new API header in that folder

Yes, sorry, I should have specified. The idea was just to create a generic, standalone driver. I have already placed it in the include/audio directory

> When writing new functionality you really want to base yourself off of the current master

Alright sure, I will do that, and create a pull request as soon as I have got it working.

frederik-popp, 2020-08-18:
I have now succeeded in creating a working sample based off of:
https://github.com/zephyrproject-rtos/zephyr/commit/9df168b53569eb29ca96cc2b8625483c8e3d9833 (Commit from August 10th 2020)

However, despite the sampling working, I can see that the file `build/zephyr/include/generated/driver-validation.h` does not include the PDM. I'm not exactly sure what this means, or how to fix it?

closed 2020-11-02 by github-actions[bot]