zephyrproject-rtos/zephyr #27463: Cannot build samples/net/sockets/echo for cc3220sf_launchxl

tbursztyka, 2020-08-10:
**Describe the bug**
Cannot build samples/net/sockets/echo for cc3220sf_launchxl

**To Reproduce**
Steps to reproduce the behavior:
1. build samples/net/sockets/echo for cc3220sf_launchxl

**Expected behavior**
build should succeed

**Impact**
Annoyance.

**Logs and console output**
```
In file included from /workdir/zephyr/subsys/net/ip/net_context.c:40: 
/workdir/zephyr/subsys/net/ip/tcp2.h:120:1: error: useless type name in empty declaration [-Werror]
 void net_tcp_init(void); 

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: (Zephyr SDK
 - Latest master

**Additional context**
It showed up in CI through PR #24873

vanti, 2020-08-10:
I have put up a fix in #27476. It is concerning however that CI didn't try to build for platforms using non-native stack when the tcp2 PR was submitted. Is this a gap that can be addressed for the future?

jukkar, 2020-08-11:
> Is this a gap that can be addressed for the future?

This should be addressed for sure. We seem to be missing tests that would validate that non-native stacks can be compiled properly. Would you be able to propose something for this?

closed 2020-08-11 by galak

vanti, 2020-08-11:
> > Is this a gap that can be addressed for the future?
> 
> This should be addressed for sure. We seem to be missing tests that would validate that non-native stacks can be compiled properly. Would you be able to propose something for this?

@jukkar I am not familiar with how CI determines which test to build given a PR, hence my question. Is this documented somewhere?

PR #27476 (2020-08-11): net: tcp2: compile out net_tcp_init() in tcp2.h for non-native stacks
For platforms using non-native stacks, net_tcp_init() should be
compiled out, similar to how it is done in tcp_internal.h.

Fixes #27463
commit 26696a4e (2020-08-11): net: tcp2: compile out net_tcp_init() in tcp2.h for non-native stacks

For platforms using non-native stacks, net_tcp_init() should be
compiled out, similar to how it is done in tcp_internal.h.

Fixes #27463