zephyrproject-rtos/zephyr #27465: How recursively build boards on Zephyr?

nandojve, 2020-08-10:
It is not clear the path about how support SIP and/or SoM on Zephyr. Many confirm that is a small board and should be added as a regular one. However, when I tried to do it I faced this: "How recursively build boards?".

It is clear that we can add a board dependency at dtsi. However, is Zephyr recursively build all dependencies?

Let's imagine this scenario: There is a SIP_FOO, based on SoC_FOO. Then, I create a module MOD_FOO based on the SIP_FOO. Then, to be useful, I create an SoM with MOD_FOO and finally the end user will create their own board Board_FOO.

The SIP needs an Pinmux and Init to configure internal buses. The Module needs the same, internal connections, PMIC etc. The SOM connects external memories and SPI WIFI, again need Pimnux and Init files. Finally, user define their own LEDs, and Sensors etc.

Is the above truly working? Is there plans to have this working?

tejlmand, 2020-08-26:
As you already seem to be aware, a SoM should be added as a regular board.
If then, someone wants to create a new board, using that SoM, they should be able to use your SoM board as a reference to extend.

An example of a SoM with a a devkit, that is based on Nordic nRF52833 can be found here:
https://github.com/zephyrproject-rtos/zephyr/tree/master/boards/arm/bl653_dvk

https://www.lairdconnect.com/wireless-modules/bluetooth-modules/bluetooth-5-modules/bl653-series-bluetooth-51-802154-nfc-module

Hope that give some inspiration.

nandojve, 2020-08-26:
Hi @tejlmand , I understand the approach that you suggested. However, I think I not made my self clear enough because there is no SoM module definition on the bl653_dvk. The board was defined same way as any chip down board itself and BL653 SoM, in fact, doesn't exists. This is not what I looking for. I'm trying to reuse/extend a SoM to create N different boards. I'm looking for a relation like we have with soc + board. Zephyr boards don't redefine SoC for every board, it extends. For instance, I can not extend BL653_dvk to create another board.

The SiP (STM32 + S2LP + FEM) could be defined as an SoC like, for instance?
http://www.htmicron.com.br/assets/files/201912_iMCP.pdf

I would like define that SiP and initialize it. After that, any user can include that and create their own board. instead copy and paste. So, any modifications on SiP itself will reflect on all derived boards as consequence. This allows us a central point of change instead replicate same changes on all variations.

Hope it is more clear :D

tejlmand, 2020-08-27:
@nandojve thanks for the detailed explanation. It's clearer now.

Unfortunately Zephyr doesn't have the concept of SoM for the time being, but this does sound like a valid enhancement in future.

The current board concept doesn't get you all the way, but there are a couple of inspirational sources that can get part of the way, and minimizing the need for copy pasting, when extending your SoM.

One part for inspiration is the non-secure versions of identical boards, such as:
nrf9160dk_nrf9160 and nrf9160dk_nrf9160ns
https://github.com/zephyrproject-rtos/zephyr/tree/master/boards/arm/nrf9160dk_nrf9160
As you see, those boards share a common: [nrf9160dk_nrf9160_common.dts](https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160_common.dts)  file:
```
#include "nrf9160dk_nrf9160_common.dts"
```
in [nrf9160dk_nrf9160](https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160.dts#L9) and [nrf9160dk_nrf9160ns.dts](https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160ns.dts#L9)

Another board to look into, is this `em_starterkit` series https://github.com/zephyrproject-rtos/zephyr/tree/master/boards/arc/em_starterkit, that does it a little different but still shares as much as possible in terms of Kconfig.board, and Kconfig.defconfig, and dts files.
There is a [boards.dtsi](https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arc/em_starterkit/board.dtsi) shared among the boards.
Or the [em_starterkit_r22.dtsi](https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arc/em_starterkit/em_starterkit_r22.dtsi) and [em_starterkit_r23.dtsi](https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arc/em_starterkit/em_starterkit_r23.dtsi) depending on the actual board.

If you notices in both examples above, the boards in a given folder share the Kconfig files.

But for both of the solutions above, then each board must still have its own:
`<board>.yaml`, even if identical to another board.yml
`<board>_defconfig` even if identical to another _defconfig.
`<board>.dts` but this board can include other dts files, and thus avoid copy pasting.

nandojve, 2020-08-28:
Hi @tejlmand, thank you clarify the current use cases. It helps me to take decisions about this topic, appreciate.

closed 2020-08-28 by nandojve

tejlmand, 2020-08-28:
> Hi @tejlmand, thank you clarify the current use cases. It helps me to take decisions about this topic, appreciate.

Happy to help.

One limitation in case downstream users of your work are to extend your SoM, is that all the solutions I pointed to extend boards in the same folder.
This could indicated that users extending your SoM could end+up having to do extension in your board (SoM) folder.
If, during your work, you see this as a limitation, and you find a nice way that SoM could be described and extended, then you are welcome to send a patch with suggestions.
Then it can be evaluated and discussed.