zephyrproject-rtos/zephyr #27471: Zephyr application build system is not aware of settings partition

MarekPieta, 2020-08-10:
**Describe the bug**
Application build system is not aware of settings partition. For big applications that may result in building improper hex. In case code does not fit on the "code partition", part of the firmware may actually reside in the settings FLASH area. 

**Expected behavior**
Developer should not be able to successfully build application that does not fit into memory (or at least there should be warning related to the problem).

**Impact**
Settings may remove/damage part of the firmware. That leads to errors that could be really difficult to debug.

nvlsianpu, 2020-09-09:
@tejlmand Can you help with this issue?

tejlmand, 2020-09-10:
@nvlsianpu sure. will take a look.

pdunaj, 2020-10-22:
Hi @MaureenHelm , I highly recommend to review the bug priority. The problem is that user may be completely unaware that part of his code is gone. It may be part of the code that is responsible for some critical action executed rarely.

pabigot, 2020-10-22:
Agreed, if the flash area used for code/data does not limit the memory region used by the linker to store code/data that's not a low priority bug.  @nashif

ioannisg, 2020-10-27:
Agreed. Changing this to medium, for now.

ioannisg, 2021-02-03:
@nashif FYI, i guess we'll release v2.5.0 with this bug - none of the proposed solutions made it into the release.

JordanYates, 2021-04-09:
Issue is not stale, solutions are being actively prototyped.

carlescufi, 2021-05-18:
@nashif @galak I wonder whether this should be an enhancement: This has always been the case, since day 1, and it doesn't prevent anyone from actually implementing a solution. Whe  using [`CONFIG_USE_DT_CODE_PARTITION`](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_USE_DT_CODE_PARTITION.html#cmdoption-arg-CONFIG_USE_DT_CODE_PARTITION) the application will fail to link if it overflows the code partition.

ioannisg, 2021-05-18:
@carlescufi how about when we do not use the code partition then?

de-nordic, 2021-05-18:
> 
> 
> @carlescufi how about when we do not use the code partition then?

Then it is not a bug because the DTS have not been used for building application so the information on application size/offset is taken from Kconfig - this is target misconfiguration. The configuration to the Kconfig, when DTS is not used, should be set with awareness of the DTS partition layout if user chooses to use both configuration methods at the same time.

This is also not "settings" partition specific but may happen to any partition. You may also apply overlay that will be dividing some partition into smaller pieces or will be defining partitions that overlap, you will not be able to detect these things during build either, and they are are target misconfigurations.

ioannisg, 2021-05-18:
> Then it is not a bug because the DTS have not been used for building application so the information on application size/offset is taken from Kconfig - this is target misconfiguration. The [...]

Then I guess, it is OK to convert this to an enhancement.

mbolivar-nordic, 2021-05-18:
I agree with @de-nordic 's assessment. If you use the settings partition in DTS in your application, but you don't instruct the linker to restrict your code partition not to overlap with the settings partition, you're "holding it wrong", although I agree the current situation makes it easier to make this mistake than it should be.

JordanYates, 2021-05-18:
I don't have a problem with this issue changing to an enhancement, but IMO this is less an issue of  "you're holding it wrong" and more an issue of "it was given to me upside down". The user doesn't need to do anything beyond turning on features for this to blow up in their face on most boards.

ioannisg, 2021-05-19:
> I agree with @de-nordic 's assessment. If you use the settings partition in DTS in your application, but you don't instruct the linker to restrict your code partition not to overlap with the [...]

The problem is that we do not have an explicit user config that signifies that we are using the DT settings partition; we only have config USE_DT_CODE_PARTITION, which only reflects the intention of using the defined code partition for the image's text area. So it's a bit unclear for the user. Furthermore, we do not warn the user that using the settings partition in the application requires using DT partitions, so, it requires setting the USE_DT_CODE_PARTITION Kconfig. Of course, we do not enforce this (automatically), either. And we do not do that, IMHO, because we admit that we have a big limitation; we do not have an efficient way of using DT partitions _without_ a bootloader, but we do wish to "use" the settings partition in applications without bootloader :). So we keep silent about it, hoping the image will not overlap with settings area, and this can only create problems for the users. That said I agree more with the "it was given to me upside down" point.

@mbolivar-nordic @de-nordic if you think it's the user's fault, would you accept something like:

```
config SETTINGS (or, any other feature that needs the storage partition, it doesn't matter)
    bool "bla bla bla: using the SETTINGS subsystem"
    select USE_DT_CODE_PARTITION
    help
       Enabling SETTINGS. This uses a defined memory region in flash for storage and flash access.
       Zephyr image cannot use this. Enable US_DT_CODE_PARTITION, to avoid this.
```

de-nordic, 2021-05-19:
I think we may need additional script that would take .config and zephyr.dts and would be run before compilaiton, and that would check if:
 1) any partition definitions (status "okay") cross each other boundaries - that would be an error;
 2) any partition is defined within range of other partition -I do not know whether there is a usecase for that, I think this is error;
 3) if DTS **is used** for code partition, if yes, we are done and OK, because previous steps ensured that there is nothing else using given range; 
 if DTS **is not used**, check if range provided in Kconfig, offset and size, do not cross into any defined partition, if yes then this is error;

That script would unfortunately disallow usage of multiple fixed-partition layouts for the same flash.

de-nordic, 2021-05-19:
@ioannisg I am fine with this, but it will just solve the "settings" problem. You may still, from within app code do something like `flash_ara_open(FLASH_AREA_ID(settings), ...)` and write over your app if "settings" overlap it.

mbolivar-nordic, 2021-05-19:
@JordanYates @ioannisg I totally agree that things are pretty messed up and hard to use right now. But I think part of the reason why is that we've evolved things organically in response to pain points and there's no overall plan, so anytime you go off of the well-beaten path, it's easy to shoot yourself in the foot.

And I take my share of blame here because I was there when the current DTS approach to partition management was added -- we definitely took shortcuts to get convenient MCUboot support into upstream zephyr without understanding DTS very well.

However, I think that `config SETTINGS` as suggested [by ioannisg](https://github.com/zephyrproject-rtos/zephyr/issues/27471#issuecomment-843838552) is only making things worse, because it's continuing the problem of ad-hoc solutions to individual user problems like "I want a storage partition, but none of the other partitions in the predefined layout, which was really designed for use with both mcuboot AND partitions".

So I'm against it and I would rather try to solve the root problem instead.

I think we should go one level up in the DT and select an entire partition layout, using something like this:

```
&flash0 {
	partition-layouts {
		partitions_mcuboot: partitions-0 {
			/*
			 * This partition layout is intended for use with
			 * mcuboot, along with an A/B application and upgrade
			 * slot with rollback support, and a storage partition.
			 */

			compatible = "zephyr,flash-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			partition@0 {
				/* mcuboot should go here */
				reg = <0x000000000 0x0000C000>;
			};
			partition@c000 {
				/* application image slot 0 */
				reg = <0x0000C000 0x00067000>;
			};
			partition@73000 {
				/* application image slot 1 */
				reg = <0x00073000 0x00067000>;
			};
			partition@da000 {
				/*
				 * scratch partition for use by mcuboot
				 * during image upgrade
				 */
				reg = <0x000da000 0x0001e000>;
			};
			partition@f8000 {
				/* storage partition for application use */
				reg = <0x000f8000 0x00008000>;
			};
		};

		partitions_storage: partitions-1 {
			compatible = "zephyr,flash-partitions";

			partition@0 {
				/* application image should go here */
				reg = <0x000000000 0x000f8000>;
			};

			partition@f8000 {
				/* storage partition for application use */
				reg = <0x000f8000 0x00008000>;
			};
		};
};

/ {
	chosen {
		/*
		 * Select the entire partitions node you want,
		 * not just a code partition.
		 */
		zephyr,flash-partitions = &partitions_mcuboot;
			/* OR */
		zephyr,flash-partitions = &partitions_storage;
			/* OR */
		zephyr,flash-partitions = &something_application_specific;
	};
};
```

Note that I've left out various details, such as:

1. How do we select the image partition within the `zephyr,flash-partitions` node? There are various options, like:
    
    - give it a magic label property, like `partitions-1 { partition@0 { label = "zephyr-image"; };`
    - add a phandle property inside the parent node, like `partitions-0 { zephyr-image-partition = &my_boot_partition; my_boot_partition: partition@foo { ... } };`
    - a Kconfig option that defaults to one of these but is user-overridable

2. How do we manage this in the linker?
3. How do we manage backwards compatibility with DTS using `fixed-partitions` compatibles and `zephyr,code-partition`?
4. What label properties and/or node labels would we want to add to this?

I think all of that needs to be discussed though. In the meantime, yes, I agree we're handing people something confusing, but I don't think that more duct tape is the right thing.

de-nordic, 2021-05-19:
> 
> I think all of that needs to be discussed though. In the meantime, yes, I agree we're handing people something confusing, but I don't think that more duct tape is the right thing.

@mbolivar You are right.

nordicjm, 2022-09-21:
Bumping a very old topic, however with an idea on top of @mbolivar-nordic's in that a useful feature would be slot selection using this method. At present, when using mcuboot with support in the application, images are build for slot0, however mcuboot does not have to operate in this mode, it can run in a direct slot execution mode whereby it does not swap the image slots but just runs the image in a particular slot directly. There is also a test of sysbuild https://github.com/zephyrproject-rtos/zephyr/pull/49552 which forces a test application to go to the secondary slot, which does not work in the case of an external memory because it is essentially a hack which just adjusts the output hex file to be an offset from the primary image. If the image slots could have groups and those groups changed depending upon what application/configuration is being created, would be a very useful improvement.