zephyrproject-rtos/zephyr #27480: remove use of ioctl() within the network stack

andrewboie, 2020-08-10:
**Is your enhancement proposal related to a problem? Please describe.**
There is an ioctl-like interface used within the networking code. I say ioctl-like but it really isn't ioctl because these functions are truly variadic instead of 2 or 3 args with the third optional arg a void pointer.

Ioctls are bad. This thread concisely explains from Linus Torvalds himself what the problems are: https://yarchive.net/comp/linux/ioctl.html

Right now, the way ioctl is used internally by networking all it does is defeat type safety. There are no advatages. We can of course keep vtable-like abstractions in the kernel, but these APIs need to have proper prototypes, see the Zephyr driver model as an example.

**Describe the solution you'd like**
Any use if ioctl() removed from the kernel with the exception of what's minimally needed for POSIX compatibility, and such code MUST be limited to lib/posix and nowhere wlse.

jukkar, 2020-08-11:
Yes, I agree with you fully. The ioctl API should be removed and the corresponding code implemented differently.

jukkar, 2025-03-07:
After revisiting this and looking the code using the ioctl, I doubt we will be easily able to extend the socket APIs to have new callbacks to support the needed use cases. Everything works atm using the ioctl so I am hesitant to touch the working code. I am not closing this if anyone wants to work on this.

JeewoongKim, 2026-09-07:
Removing the ioctl uses in one change would make the scope fairly large, so I started with a small PR that converts the core socket poll paths incrementally.

I'd like to get feedback on this approach in #118426 first, and then continue converting the remaining users in follow-up PRs if it looks reasonable.