zephyrproject-rtos/zephyr #27484: sanitycheck: Ease error interception from calling script

erwango, 2020-08-11:
**Is your enhancement proposal related to a problem? Please describe.**
Today it is not easy to detect sanitycheck execution errors from a calling script.
Some non error log is redirected to stderr, so for instance the following command:
```
./scripts/sanitycheck --ninja --runtime-artifact-cleanup -T samples/hello_world/ -p nucleo_f401re --outdir /local/mcu/zephyrproject/logs/200811/log_db_master_9df168b535_0950 2> error.tmp
```
outputs:
```
Renaming output directory to /local/mcu/zephyrproject/logs/200811/log_db_master_9df168b535_0950.5
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    0
```
And output the 2 following lines in errror.tmp:
```
INFO    - JOBS: 20
INFO    - Building initial testcase list...
INFO    - 1 test configurations selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - 1 of 1 tests passed (100.00%), 0 failed, 0 skipped with 0 warnings in 3.97 seconds
INFO    - In total 1 test cases were executed on 1 out of total 272 platforms (0.37%)
INFO    - 0 tests executed on platforms, 1 tests were only built.
```

**Describe the solution you'd like**
Normalize sanitycheck script output so only script execution errors are dumped to stderr and sanitycheck callers can detect script execution issues.

**Additional context**
Identifying script execution errors is required to ease daily test bench maintenance

nashif, 2020-11-28:
this could be resolved with:
```diff
diff --git a/scripts/sanitycheck b/scripts/sanitycheck
index 16453aa052..19a0949020 100755
--- a/scripts/sanitycheck
+++ b/scripts/sanitycheck
@@ -763,7 +763,7 @@ def main():
     fh.setLevel(logging.DEBUG)

     # create console handler with a higher log level
-    ch = logging.StreamHandler()
+    ch = logging.StreamHandler(sys.stdout)

     VERBOSE = options.verbose
     if VERBOSE > 1:
```

however, what type of errors you expect to go to stderr?

erwango, 2020-11-30:
> however, what type of errors you expect to go to stderr?

Good question, this was some time ago, I should have documented.
So, I can recall exactly but these were script execution issues that could be due to wrong environment setup or alike. Not test execution issues, but kind of errors that would lead to wrong test results.
Though, fact that I am not able to recall is a sign that this was not that critical or that I managed to get through it.

So either the fix is harmless and let's add it, either we can close the point and I'll re open once I face one of these issues once again

erwango, 2021-04-26:
Closing and will re-open once the case is faced again

closed 2021-04-26 by erwango