zephyrproject-rtos/zephyr #27488: Bluetooth Mesh sample don't work

MajdaKHN, 2020-08-11:
**Describe the bug**
I'm using the nrf52dk_nrf52832 board and I develop Bluetooth Mesh devices. I update Zephyr to 2.3 version and tried to build Bluetooth mesh sample in :
> sample/bluetooth/mesh
> sample/bluetooth/mesh_demo

And none of them build.

**To Reproduce**
Steps to reproduce the behavior:
1. cd sample/bluetooth/mesh
2. west build -p auto -b nrf52dk_nrf52832

**Expected behavior**
build

**Impact**
don't build

**Logs and console output**
Console log of the build :

```
C:\Zephyr\zephyr_2-3\zephyr\zephyrproject\zephyr\samples\bluetooth\mesh_demo>west build -p auto -b nrf52dk_nrf52832
[114/198] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj
C:\gnuarmemb\bin\arm-none-eabi-gcc.exe -DBUILD_VERSION=zephyr-v2.3.0-1685-gb143213dc1ea -DKERNEL -DNRF52832_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -IC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/include -Izephyr/include/generated -IC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52 -IC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/subsys/settings/include -IC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/subsys/bluetooth -IC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/subsys/bluetooth/controller/ll_sw/nordic -IC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -IC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/modules/hal/nordic/nrfx -IC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/modules/hal/nordic/nrfx/drivers/include -IC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/modules/hal/nordic/nrfx/mdk -IC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/modules/hal/nordic/. -IC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/modules/debug/segger/rtt -IC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/modules/crypto/tinycrypt/lib/include -isystem C:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/lib/libc/minimal/include -isystem c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/7.3.1/include -isystem c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/7.3.1/include-fixed -Os -imacrosC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/samples/bluetooth/mesh_demo/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacrosC:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles\app.dir\src\main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj   -c ../src/main.c
In file included from C:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/include/bluetooth/mesh.h:17:0,
                 from ../src/main.c:14:
C:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/include/bluetooth/mesh/access.h:20:4: warning: implicit declaration of function 'UTIL_LISTIFY'; did you mean 'UTIL_IIF'? [-Wimplicit-function-declaration]
  { UTIL_LISTIFY(CONFIG_BT_MESH_MODEL_KEY_COUNT, \
    ^
C:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/include/bluetooth/mesh/access.h:21:10: error: 'BT_MESH_KEY_UNUSED_ELT_' undeclared here (not in a function)
          BT_MESH_KEY_UNUSED_ELT_) }
          ^
C:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/include/bluetooth/mesh/access.h:24:10: error: 'BT_MESH_ADDR_UNASSIGNED_ELT_' undeclared here (not in a function); did you mean 'BT_MESH_KEY_UNUSED_ELT_'?
          BT_MESH_ADDR_UNASSIGNED_ELT_) }
          ^
C:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/include/bluetooth/mesh/access.h:20:4: error: initializer element is not constant
  { UTIL_LISTIFY(CONFIG_BT_MESH_MODEL_KEY_COUNT, \
    ^
C:/Zephyr/zephyr_2-3/zephyr/zephyrproject/zephyr/include/bluetooth/mesh/access.h:23:4: error: initializer element is not constant
  { UTIL_LISTIFY(CONFIG_BT_MESH_MODEL_GROUP_COUNT, \
    ^
[123/198] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/fdtable.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\CMake\bin\cmake.EXE' --build 'C:\Zephyr\zephyr_2-3\zephyr\zephyrproject\zephyr\samples\bluetooth\mesh_demo\build'
```

Some `note` are in the console log, I removed it of the issue, but they are on the access.h too.

**Environment :**
 - OS: Windows 10
 - Zephyr 2.3.99
 - cmake version 3.13.2
- West version: v0.7.3

**Additional information :**
I tried to build with different board like th bbc_microbit or nrf51_blenano but none of them build.

MajdaKHN, 2020-08-11:
The console log is for `mesh_demo` sample, for the `mesh` sample, I just have the two first error

MajdaKHN, 2020-08-11:
I cloned another zephyrproject and no more error occured.
I tried to fix this bug with `west` command but with no success.
But with the bugged zephyr version i had error on `west update` : 
`ERROR: update failed for project hal_nxp`
But I don't think it's related

carlescufi, 2020-08-11:
@MajdaKHN I just built both samples for nrf52dk_nrf52832 both on Zephyr 2.3.0 and the current master and both build fine. This must be an issue with your build system. Please try deleting your `build/` folder entirely.

closed 2020-08-11 by carlescufi