zephyrproject-rtos/zephyr #27490: arch/common/isr_tables.c compilation fails with CONFIG_NUM_IRQS=0

henrikbrixandersen, 2020-08-11:
**Describe the bug**
The `isr_tables.c` file fails to compile with CONFIG_NUM_IRQS=0. This was found when reviewing #27394 along with @ioannisg.

**To Reproduce**
Steps to reproduce the behavior:
0. Change a SoC definition to set `CONFIG_NUM_IRQS=0`
1. cd samples/hello_world
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz -DCONFIG_UART_INTERRUPT_DRIVEN=n
3. make
4. See error:

```
Scanning dependencies of target arch__common
[ 56%] Building C object zephyr/arch/common/CMakeFiles/arch__common.dir/isr_tables.c.obj
/home/brix/Projects/zephyrproject/zephyr/arch/common/isr_tables.c:54:3: error: array index in initializer exceeds array bounds
   54 |  [0 ...(IRQ_TABLE_SIZE - 1)] = (uint32_t)&IRQ_VECTOR_TABLE_DEFAULT_ISR,
      |   ^
/home/brix/Projects/zephyrproject/zephyr/arch/common/isr_tables.c:54:3: note: (near initialization for '_irq_vector_table')
/home/brix/Projects/zephyrproject/zephyr/arch/common/isr_tables.c:54:32: warning: excess elements in array initializer
   54 |  [0 ...(IRQ_TABLE_SIZE - 1)] = (uint32_t)&IRQ_VECTOR_TABLE_DEFAULT_ISR,
      |                                ^
/home/brix/Projects/zephyrproject/zephyr/arch/common/isr_tables.c:54:32: note: (near initialization for '_irq_vector_table')
/home/brix/Projects/zephyrproject/zephyr/arch/common/isr_tables.c:63:3: error: array index in initializer exceeds array bounds
   63 |  [0 ...(IRQ_TABLE_SIZE - 1)] = {(void *)0x42, (void *)&z_irq_spurious},
      |   ^
/home/brix/Projects/zephyrproject/zephyr/arch/common/isr_tables.c:63:3: note: (near initialization for '_sw_isr_table')
/home/brix/Projects/zephyrproject/zephyr/arch/common/isr_tables.c:63:32: warning: excess elements in array initializer
   63 |  [0 ...(IRQ_TABLE_SIZE - 1)] = {(void *)0x42, (void *)&z_irq_spurious},
      |                                ^
/home/brix/Projects/zephyrproject/zephyr/arch/common/isr_tables.c:63:32: note: (near initialization for '_sw_isr_table')
zephyr/arch/common/CMakeFiles/arch__common.dir/build.make:82: recipe for target 'zephyr/arch/common/CMakeFiles/arch__common.dir/isr_tables.c.obj' failed
make[2]: *** [zephyr/arch/common/CMakeFiles/arch__common.dir/isr_tables.c.obj] Error 1
CMakeFiles/Makefile2:2403: recipe for target 'zephyr/arch/common/CMakeFiles/arch__common.dir/all' failed
make[1]: *** [zephyr/arch/common/CMakeFiles/arch__common.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2

```

**Expected behavior**
Compilation succeeds.

**Impact**
It is not possible to support an SoC without IRQs implemented.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: 2505daf28f378aa968cd4e4a7fbdeaaf3eeeeaa8

ceolin, 2020-09-22:
You can achieve that with `CONFIG_GEN_IRQ_VECTOR_TABLE=n`. When this option is enabled (that is the common case) a lot of other places assume that CONFIG_NUM_IRQS is greater than 0. I may be no understanding the problem but I think that if you have NUM_IRQS = 0 you don't want to a vector table generated

ceolin, 2020-09-23:
> You can achieve that with `CONFIG_GEN_IRQ_VECTOR_TABLE=n`. When this option is enabled (that is the common case) a lot of other places assume that CONFIG_NUM_IRQS is greater than 0. I may be no [...]

@henrikbrixandersen

closed 2020-09-29 by ioannisg