zephyrproject-rtos/zephyr #27495: Include full register state in ARM Cortex M Exception Stack Frame (ESF)

chrisc11, 2020-08-11:
**Is your enhancement proposal related to a problem? Please describe.**

To debug hard-to-reproduce faults/panics, it's helpful to get the full
register state at the time the fault occurred. This enables recovering
full backtraces and the state of local variables at the time a fault
took place.

Pretty much all OSs and even some RTOS (i.e esp-idf) support this in
some fashion (i.e coredumps, mini-dumps, linux kernel oops, etc) and
it would be awesome if Zephyr could too!

Today this is not possible with the ARM Cortex M Zephyr port for two
main reasons:

1. A [copy of the original hardware exception stack frame](https://github.com/zephyrproject-rtos/zephyr/blob/a1b77fd589dbe7284c17b029f251426a724abd47/arch/arm/core/aarch32/cortex_m/fault.c#L961-L962)
is made before calling k_sys_fatal_error_handler(). This means the
stack pointer prior to the exception cannot be readily resolved making
it impossible to recover the full backtrace.

2. Registers that are callee-saved (r4-r11 & exc_return) are not
collected in z_arm_fault() which means when k_sys_fatal_error_handler
is called the register state is not readily available. This
information is needed to accurately recover local variable state
leading up to a crash.

**Describe the solution you'd like**

To fix these issues and enable better diagnostic facilities, I'd like
to update the [`z_arch_esf_t` implementation for ARM aarch32](https://github.com/zephyrproject-rtos/zephyr/blob/a1b77fd589dbe7284c17b029f251426a724abd47/include/arch/arm/aarch32/exc.h#L76-L92)
to pass the original hardware exception frame pointer as well as
callee-saved registers.

```c
struct __full_esf {
	struct __esf *exception_frame;
	struct __callee_saved_esf *callee_regs;
};

typedef struct __full_esf z_arch_esf_t;
```

**Describe alternatives you've considered**

There are multiple ways the full stack frame could be exposed from
z_arm_fault() (new params in the fault handler, a cortex-m specific
api, etc). I ultimately opted to use an approach that would allow the
information to be exposed via the current k_sys_fatal_error_handler
API.

**Additional context**

I've been using local patches to achieve this behavior on the last few
Zephyr releases but I'd like to upstream the work so it's more
accessible to others. I've put together a working implementation [on this branch  for reference](https://github.com/chrisc11/zephyr/tree/ccoleman/full_esf).

galak, 2020-08-11:
@chrisc11 might be good to submit this as a draft PR.

chrisc11, 2020-08-11:
Thanks for the suggestion @galak. I just submitted a draft PR

ioannisg, 2020-08-31:
Fixed by #27496

closed 2020-08-31 by ioannisg