zephyrproject-rtos/zephyr #27505: spi: mchp: Unintended data is transmitted when tx and rx operations are performed simultaneously

albertofloyd, 2020-08-12:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. modify helloworld to use SPI driver
2. west build -b mec15xxx
3. make

Power on while capturing SPI traffic. 

SPI lines show more traffic than intended .
SDO = 0x05 , 0xFF, 0x00, 0x03
SDI  = 0xFF, 0x00, 0x00, 0x00

**Expected behavior**
A clear and concise description of what you expected to happen.

Expected tx traffic should be only 2 bytes ( 0x05 + dummy opcode ).
 
SDO = 0x05 , 0xFF
SDI  = 0xFF, 0x00

**Impact**
Medium, this behavior could lead to unintended behavior.

**Logs and console output**
struct spi_cmd {
	u8_t buf[10];
	u8_t tx_len;
	u8_t rx_len;
};

static struct spi_cmd cmds[] = {
	[RD_STS1_CMD_INDEX] = {
		.buf = { READ_STATUS_1_OPCODE, DUMMY_OPCODE },
		.tx_len = 2,
		.rx_len = 2,
	},
};

int spi_send_cmd(u8_t slave_index, struct spi_cmd *cmd)
{
	u8_t data[10];
	int ret;

	tx.buf = cmd->buf;
	tx.len = cmd->tx_len;

	rx.buf = data;
	rx.len = cmd->rx_len;

	LOG_DBG("Sending cmd %x SPI flash device %x", cmd->buf[0], slave_index);
	LOG_DBG("Tx len %x rx len %x", cmd->tx_len, cmd->rx_len);
	spi_cfg.slave = slave_index;
	ret = spi_transceive(spi_dev, &spi_cfg, &tx_bufs, &rx_bufs);
	if (ret < 0) {
		LOG_ERR("SPI transceive error: %d", ret);
		return ret;
	}
}

spi_send_cmd(slave_index, &cmds[RD_STS1_CMD_INDEX]);

**Additional context**
Add any other context about the problem here.

scottwcpg, 2020-08-12:
I'm not reproducing this. Maybe my test code is different enough.  Attached my main.c and a scope capture showing only 0x05,0xFF transmitted and 0x00,0x00 read.  W25Q128 Status 1 register defaults to 0x00.
[image]
[main.zip](https://github.com/zephyrproject-rtos/zephyr/files/5065180/main.zip)

scottwcpg, 2020-08-12:
Note: the QMSPI hardware design is different than many other SPI controllers.  While transmitting data it does not capture data on input pins.  This means only the opcode and optional parameters should be in the TX buffer passed to the driver.  There is no need to tell the driver to manually generate clocks by transmitting extra data.  The receive length passed to the driver configures the HW to generate the required number of clocks for data read.

albertofloyd, 2020-08-12:
Correct, problem is in the test just as you described.

closed 2020-08-12 by albertofloyd