zephyrproject-rtos/zephyr #27510: Bluetooth: controller: v1.14.x - conn-ind with an all zero chmap crash

erbr-ot, 2020-08-12:
When a connect indication contains a channel map of all zeros, the adv->conn is left NULL'ed after connect establishment is abborted and next connect attempt leads to a crash.

This was identified via the Sweyntooth test suite - executing the script re. issue 6.14 towards an Oticon target.
https://asset-group.github.io/disclosures/sweyntooth/

See #27507

carlescufi, 2020-09-15:
@erbr-ot will you create a backport PR for this, targetting the v1.14-branch?

erbr-ot, 2020-09-16:
@carlescufi I will, in the next few days.

erbr-ot, 2020-10-13:
Hi @carlescufi - sorry this got back-seated during our race to finish-line. Looking into this now, and have a question: I see that the guard re. lll->data_chan_hop is not present in v1.14.2, should I introduce this also in the same PR?

carlescufi, 2020-10-19:
> Hi @carlescufi - sorry this got back-seated during our race to finish-line. Looking into this now, and have a question: I see that the guard re. lll->data_chan_hop is not present in v1.14.2, should I introduce this also in the same PR?

I think this is a question for @cvinayak

closed 2021-01-02 by github-actions[bot]

cvinayak, 2021-03-29:
This slipped through me. Now addressed here https://github.com/zephyrproject-rtos/zephyr/pull/33760